foxglove/studio #4689: Can't open UDP port for Velodyne LIDAR on the V1.29.1

jessaggie, 2022-10-21:
**Description**
When trying to open the UDP port connection to the Velodyne Lidar, nothing happens when pressing open. 

- Version:
- OS:  Windows 10
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):  Velodyne Lidar connection
- Foxglove version V1.29.1

**Steps To Reproduce**

1. Add a data source
2. click open connection
3. click Velodyne LIDAR
4. type in any port
5. press open

**Expected Behavior**
At this point, the port should open and the open new connection window should close

**Actual Behavior**
The open new connection window does not close and nothing happens when you click open.

defunctzombie, 2022-10-21:
Are you able to open the Velodyne management webpage in your web browser?

Is your wired network configured correctly (with the correct IP and subnet)?

amacneil, 2022-10-24:
@jtbandes: When the lidar is not reachable we never close the dialog making it look like we never handled your click on the "open" button?

@defunctzombie: I will check this behavior.

PR #4728 (2022-11-01): Emit state during velodyne player setup

**User-Facing Changes**
When connecting to Velodyne data sources, the open dialog is immediately dismissed upon clicking "open".

**Description**
When selecting a data source, the open dialog remains open until a player presence other than NOT_PRESENT is set of the player state. Since the velodyne player would only produce a player state when receiving its first message the open dialog would remain open after setting the port and clicking "open". This behavior makes it look like the dialog is broken and the user's click action was not registered.

This  [... cut]
commit ec8f1f74 (2022-11-01): Emit state during velodyne player setup (#4728)

When selecting a data source, the open dialog remains open until a
player presence other than NOT_PRESENT is set of the player state. Since
the velodyne player would only produce a player state when receiving its
first message the open dialog would remain open after setting the port
and clicking "open". This behavior makes it look like the dialog is
broken and the user's click action was not registered.

This change updates the Velodyne player to emit a state update during
initialization which clears the open dialog and shows the user the data
s [... cut]
closed 2022-11-01 by defunctzombie