foxglove/studio #4694: Log panel caches messages from previous data source

jtbandes, 2022-10-21:
The Log panel keeps some stale data around when switching data sources. In this screen recording, the first bag has log messages, but the second does not. Yet when I drop the second bag, I still see log messages from the first bag. 😱 

https://user-images.githubusercontent.com/14237/197283491-0ad1c939-a4f3-4297-abb8-2ed42f9e8ada.mov

foxymiles, 2022-10-24:
I think this might be player related. If I log what messages the panel receives from `useMessagesByTopic` during the transition it first receives an empty array, then a fresh array containing one message from the previous datasource.

It also seems as if the message reducer is being called at regular intervals even though the data source is a static file and playback is stopped.

Any ideas @defunctzombie? Is that the behavior you'd expect for a static source?

jtbandes, 2022-10-24:
> It also seems as if the message reducer is being called at regular intervals even though the data source is a static file and playback is stopped.

I've seen this before in other contexts; I believe this is an issue with the iterable player:

https://github.com/foxglove/studio/blob/cccc3a9eca94b91db146bc7418c6b80892ff006a/packages/studio-base/src/players/IterablePlayer/IterablePlayer.ts#L888-L891

closed 2022-10-25 by jtbandes