foxglove/studio #4697: rosbridge crashes when using with a ROS2 application

kevin-kindred, 2022-10-22:
**Description**

- Version: Version 1.28.0 (darwin)
- OS: MacOS 12.6
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): rosbridge
- ROS distro (if applicable): ROS2 humble

**Steps To Reproduce**
1. Started a rosbridge using 
```
apt update; apt install -y ros-humble-rosbridge-server
source /opt/ros/humble/setup.bash;
ros2 launch rosbridge_server rosbridge_websocket_launch.xml
```
2. Opened a connection to port 9090
3. Start a ROS2 application that uses moveit

**Expected Behavior**
I see the STL meshes loaded in the 3D panel

**Actual Behavior**
Nothing gets loaded & I see the rosbridge crash with the following logs
```
[rosbridge_websocket-1] ERROR:tornado.application:Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0x7fe04832d720>>, <Future finished exception=WebSocketClosedError()>)
[rosbridge_websocket-1] Traceback (most recent call last):
[rosbridge_websocket-1]   File "/usr/lib/python3/dist-packages/tornado/ioloop.py", line 741, in _run_callback
[rosbridge_websocket-1]     ret = callback()
[rosbridge_websocket-1]   File "/usr/lib/python3/dist-packages/tornado/ioloop.py", line 765, in _discard_future_result
[rosbridge_websocket-1]     future.result()
[rosbridge_websocket-1]   File "/usr/lib/python3/dist-packages/tornado/gen.py", line 234, in wrapper
[rosbridge_websocket-1]     yielded = ctx_run(next, result)
[rosbridge_websocket-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosbridge_server/websocket_handler.py", line 197, in prewrite_message
[rosbridge_websocket-1]     future = self.write_message(message, binary)
[rosbridge_websocket-1]   File "/usr/lib/python3/dist-packages/tornado/websocket.py", line 337, in write_message
[rosbridge_websocket-1]     raise WebSocketClosedError()
[rosbridge_websocket-1] tornado.websocket.WebSocketClosedError
[rosbridge_websocket-1] ERROR:tornado.application:Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0x7fe04832d720>>, <Future finished exception=WebSocketClosedError()>)
[rosbridge_websocket-1] Traceback (most recent call last):
[rosbridge_websocket-1]   File "/usr/lib/python3/dist-packages/tornado/ioloop.py", line 741, in _run_callback
[rosbridge_websocket-1]     ret = callback()
[rosbridge_websocket-1]   File "/usr/lib/python3/dist-packages/tornado/ioloop.py", line 765, in _discard_future_result
[rosbridge_websocket-1]     future.result()
[rosbridge_websocket-1]   File "/usr/lib/python3/dist-packages/tornado/gen.py", line 234, in wrapper
[rosbridge_websocket-1]     yielded = ctx_run(next, result)
[rosbridge_websocket-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosbridge_server/websocket_handler.py", line 197, in prewrite_message
[rosbridge_websocket-1]     future = self.write_message(message, binary)
[rosbridge_websocket-1]   File "/usr/lib/python3/dist-packages/tornado/websocket.py", line 337, in write_message
[rosbridge_websocket-1]     raise WebSocketClosedError()
[rosbridge_websocket-1] tornado.websocket.WebSocketClosedError
```

amacneil, 2022-10-24:
@defunctzombie @achim-k we should create a bug for this in the rosbridge repo right?

achim-k, 2022-10-26:
> @defunctzombie @achim-k we should create a bug for this in the rosbridge repo right?

Yes.

@kevin-kindred can you open an issue on the [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) repo? From the error logs, it seems like https://github.com/RobotWebTools/rosbridge_suite/issues/820 might be related.

defunctzombie, 2022-10-27:
Closing this issue in favor of an existing or new issue being created in the rosbridge_suite repo.

closed 2022-10-27 by defunctzombie

kevin-kindred, 2022-10-27:
I see that there's a new release for rosbridge-server(Oct 21 2022). I'll try that before opening an issue on the rosbridge_suite repo. Thanks guys!