foxglove/studio #4699: Visualization_msgs/Marker cube_list not updating correctly

wouter-heerwegh, 2022-10-24:
When using a cube_list as visualization_msgs/Marker, the cubes do not get updated correctly in the 3D panel. They keep the position of the first position that is published. When looking at the raw data, the cubes do get different coordinates.

- Version: 1.29.1
- OS: Ubuntu 20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): both native ROS1 and rosbag
- ROS distro (if applicable): Noetic

**Steps To Reproduce**

- Download [this bag]()
- Subscribe to the `/conestreamer/ground_truth_track`
- Show this topic in the 3D panel

**Expected Behavior**
Constantly updating positions of the cubes

**Actual Behavior**
Cubes not moving, some are disappearing and coming back in the same position.

snosenzo, 2022-10-25:
Hey, just landed a fix. Should be out in the next release! Thanks for filing and let us know if you have any more issues!

closed 2022-10-25 by snosenzo

wouter-heerwegh, 2022-10-26:
Tested with 3089eadac185f934ea6bc37c890fe11080cefd5b, works correctly. Thanks for the fix!

PR #4705 (2022-10-25): Fix Cube List markers not updating
**User-Facing Changes**
 - fix cube list (type = 6) marker primitives not updating

**Description**
 - attributebuffers need `needsUpdate = true` after they're changed to be sent to the GPU again. should also fix similar issue with spheres and other `DynamicInstanceMesh` renderables

Fixes #4699
commit 39bbe37e (2022-10-25): Fix Cube List markers not updating (#4705)

**User-Facing Changes**
 - fix cube list (type = 6) marker primitives not updating

**Description**
- attributebuffers need `needsUpdate = true` after they're changed to be
sent to the GPU again. should also fix similar issue with spheres and
other `DynamicInstanceMesh` renderables

Fixes #4699