foxglove/studio #4703: Map panel: Display GPS position history

ysuurmeijer, 2022-10-25:
Currently the map panel only displays the latest GPS position when using it for ROSbag replay. It would be to have support for analyzing historic trajectories based on a LocationFix topic. 

- Version: v1.29.0
- OS: Ubuntu (docker)
- Data source: rosbridge
- ROS distro (if applicable):

**Current Experience**
GPS map plot only shows the latest GPS position when replaying data from a ROS bag over ROS bridge which makes it very difficult to analyse trajectories.

**Desired Experience**
Ability to view full trajectory from a live GPS topic. e.g. Add an option to keep 'historic' GPS points plotted on the map, either by setting a flag for the entire plot, or for instance by adding a parameter for each series to keep N historic points plotted on the map.

jtbandes, 2022-10-25:
Duplicate of #2797

jtbandes, 2023-02-08:
I am reopening this issue since it occurs to me that #2797 was about preloading data, whereas this issue is about live connections. We should keep a history of points for live connections (like we do for the Plot panel).