foxglove/studio #4704: Is internet connection needed during ROS2 initial connection?

felixf4xu, 2022-10-25:
When I tried to open a ROS2 connection, if the PC has no internet connection, the data source side bar will display in-progress state forever. `ros2 topic list` and `ros2 topic echo` works as expected at that moment.

If the pc has internet connection, it works ok.

- Version: 1.25
- OS: Ubuntu 20
- Data source : ROS 2
- ROS distro (if applicable): Galactic

jtbandes, 2022-10-25:
Are you using the desktop app or the web app? You will need the desktop app for offline use.

jhurliman, 2022-10-25:
This is caused by https://github.com/foxglove/rtps/issues/17

felixf4xu, 2022-10-26:
Nice to know that, I will come back if the walkaround (disabling all other network adapters) works and come back.

jtbandes, 2022-11-30:
Going forward, we plan to deprecate and remove the ROS 2 native connection (https://github.com/foxglove/studio/issues/4900). We are working on a new [foxglove_bridge](https://github.com/foxglove/ros-foxglove-bridge), which will provide improved performance and reliability over the Rosbridge and native connections. You can already use the new bridge today (see the README). Stay tuned for upcoming announcements about the new bridge!

closed 2022-11-30 by jtbandes