foxglove/studio #4711: Clear/Preserve State based on WS session id

banisadr, 2022-10-26:
Depends on: https://github.com/foxglove/ws-protocol/issues/256

Need to clear or preserve state in studio when using WS connection based on session id. If reconnecting to the same session, then state should be preserved because we are recovering a dropped connection. If session id changes, we should clear state to properly display data.

Context: Customer is reconnecting to a websocket in a different context and state is being preserved even though a different program is actually being run. Its results in broken and confusing results in studio. It is not sufficient to always clear state on new connection because we may be re-connecting a dropped connection in which case state should be preserved.

jtbandes, 2022-10-26:
Currently, state will be cleared when the receiveTime of incoming messages goes backwards. I recommend the customer try sending a message with time zero when the connection needs to be reset.

amacneil, 2022-10-27:
@jtbandes do you know if we respect and clear state if the `/run_id` parameter changes?

jtbandes, 2022-10-27:
Definitely not with the foxglove websocket since it doesn’t support params 😀 
Also I don’t believe we use /run_id in the ROS native players either.