foxglove/studio #4722: VariableSidebar retaining stale globalVariables state

invizibility, 2022-10-28:
**Description**

- Version: 1.29.1
- OS: OSX Monterrey 
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): N/A
- ROS distro (if applicable): N/A

**Steps To Reproduce**

I'm working on some new control panels that let me update `globalVariables` but I'm running into some strange behavior with the `JsonInput` component inside the `VariableSidebar` . When I create and initialize a new variable in the sidebar and then attempt to modify it with my control panel, the JsonInput is showing a wrong value even though i get the green flash indicating a variable has been selected/updated.

I even added traces on line 54 of `index.tsx` in the useEffect in the `VariableSidebar` component, line 332 of `Variable.tsx`, as well as a useEffect in line 79 of `JsonInput.tsx`, all of which are showing the state of globalVariables and even the specific value prop in question is being updated, but the UI is not showing this update.

If i close the whole variable sidebar and re-open it, thereby forcing a re-render of all the components inside the sidebar, the value is displayed correctly and will subsequently obey updates from my control panel. Is there something I'm missing about `JsonInput `or `VariableSidebar` that is key to getting them to re-render correctly in this case?

**Expected Behavior**

I expected the update to the global variable to propagate correctly into the `VariableSidebar` component

**Actual Behavior**

https://user-images.githubusercontent.com/4536381/198748443-b017c0f8-0105-4436-97b4-2e5c53569519.mov

foxymiles, 2022-10-30:
Another way to reproduce this:

1. Create a new layout.
2. Add a variable slider.
3. Create a new variable in the sidebar called `globalVariable`.
4. Notice that moving the slider updates the value.
5. Delete the value from the sidebar so the variable is still there but has an empty value.
6. Move the slider.

Notice that the value no longer updates.

foxymiles, 2022-10-30:
Simpler repro:

1. Add a new variable in the sidebar.
2. Type in a new value.
3. Adjust it with the slider.

Nothing happens.

closed 2022-10-31 by defunctzombie