foxglove/studio #4729: WebRTC as a streaming data source

allsey87, 2022-11-01:
I would like to stream data into Foxglove (running in the browser) from another webapp. I think the ideal technology to achieve this is WebRTC which would enable Foxglove and the webapp to exchange data in a P2P manner, without sending it via your servers (although some server infrastructure may be required for signalling and establishing the connection). Any thoughts?

jtbandes, 2022-11-03:
> would enable Foxglove and the webapp to exchange data in a P2P manner, without sending it via your servers

This is already possible with the [Foxglove WebSocket and Rosbridge connections](https://foxglove.dev/docs/studio/connection/data-sources). Have you given those a try?

allsey87, 2022-11-03:
I don't think this would enable P2P communication between two webapps. Where would the WebSocket server run?

jtbandes, 2022-11-03:
You're right, it would not work for a web page to communicate with another web page. Can you describe more about your use case and what the other web app is?

jtbandes, 2022-11-03:
See also: https://github.com/foxglove/studio/discussions/4613#discussioncomment-3864650

allsey87, 2022-11-03:
Some game engines that are being used for alternative simulators can be compiled to Webassembly and run directly in the browser. It would be neat if you could have a simulation running in one window and have the data from the simulator forwarded to Foxglove in another window without installing anything locally.

allsey87, 2022-11-03:
Regarding UDP, it is not possible to open a UDP connection using standard web APIs.