foxglove/studio #4732: FrameTransforms not detected properly when using live websocket connection

pickledgator, 2022-11-01:
**Description**
- Version: 1.29 and 1.30
- OS: MacOS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Live websocket stream using protobuf API
- ROS distro (if applicable): 

**Steps To Reproduce**
Send a `foxglove.FrameTransport` protobuf message via websocket API

**Expected Behavior**
Client detects the transform properly, as it did in version 1.28
[image]

**Actual Behavior**
3D panel (and potentially others) does not detect the transform
Some panels (raw data) detect transform correctly

[image]
[image]

jtbandes, 2022-11-10:
I tested FrameTransform over WS with a simple test server: https://gist.github.com/jtbandes/9e738c11bb8ec9ba36536837e7990f17

It seems to work okay, so waiting for more user feedback.

https://user-images.githubusercontent.com/14237/200989163-cc62f716-a45c-4635-81ab-8b1dddb714c0.mov

pickledgator, 2022-11-15:
> @pickledgator <https://github.com/pickledgator> Do you have more steps or
> context to add for us to reproduce this? I also recommend opening the "data
> source info" sidebar (the item with the red dot) and seeing what the
> problem is. This might explain why you are not seeing the frame.

I’m traveling for the next week or so and then will jump back on this.
Sorry for the delay in feedback.

defunctzombie, 2022-11-15:
@pickledgator No worries. We'll keep this alive waiting for your response.

maxgaw, 2022-11-18:
@defunctzombie @jtbandes  I can jump in since I have the same issue. It seems like the 3D panel is not able to use the FrameTransform when using a Websocket connection.
In the video you can see the live connection first. At 00:21 you can see the exact same data loaded from the data-portal. 
Besides the message missing frame to frame, nothing else gets reported. In this example the frame hierarchy should be root > lane_frame > host_frame - mov_obj_frame as seen in the example at 00:21
When using build 1.28 everything works as expected.
https://user-images.githubusercontent.com/73784937/202665785-b73260a6-2543-42c6-86b0-d43f6aa93cfc.mp4

acowley, 2022-11-18:
I believe I’m seeing this, too. The Legacy 3D panel is picking up all the transforms, while the new one is not with a websocket connection. If I bridge things to ROS on my side, and have Foxglove connect to ROS, then the new panel works.

jtbandes, 2022-11-18:
Can someone who's able to reproduce the issue please create a modified version of this script to demonstrate it? https://gist.github.com/jtbandes/9e738c11bb8ec9ba36536837e7990f17

I'm still not able to reproduce any problems. I tried including multiple FrameTransform topics like @maxgaw's example video showed, but it still works fine. The publish rate I used is also higher.

https://user-images.githubusercontent.com/14237/202771125-a1bc01ad-4cdf-464c-8390-23a247632f88.mov

maxgaw, 2022-11-18:
@jtbandes I just tried it. It works fine. Looking at your script I dont see any differences from what I am doing, besides not building the FileDescriptorSet in the script. I am importing the file generated by protoc

jtbandes, 2022-11-18:
Maybe there is some more subtle difference like the order of messages, delays between messages, tree hierarchy, publish frequency, etc. If you could try making the actual data published more similar to your data that might help. Alternatively you can start with your non-working server, and modify it to strip away unnecessary parts one by one.

I don't think the FileDescriptorSet thing would make any difference. You can try `pip3 install mcap-protobuf-support` and `from mcap_protobuf.schema import build_file_descriptor_set` if you want to try using that in your own server.

jtbandes, 2022-12-05:
Closing out this issue for now as we haven't been able to reproduce the problem. Still happy to reopen and look into it if someone is able to share reproduction steps.

closed 2022-12-05 by jtbandes

pickledgator, 2022-12-09:
Finally got around to testing this out with the provided python script and I believe I found the discrepancy. Prior to version 1.29, in the channel advertisement, if you provided the `schemaName` as simply the message name, things worked fine. In 1.29, it seems there was a breaking change that now requires the `schemaName` to now include the full proto path. Ie, Previously `FrameTransform` was sufficient. But start in 1.29, the schemaName needs to be `foxglove.FrameTransform`. Only the 3D panel usage seemed to be affected (ie, the FrameTransform (and other message types still worked fine in the RAW panel). This is a quick fix on our end, since the proto descriptor has both `name()` and `full_name()` available to choose from. Hope this helps others.

jtbandes, 2022-12-09:
Hm. That's odd since the change you describe is what I expect from this PR, which hasn't merged yet: https://github.com/foxglove/studio/pull/4948 I guess I will try and understand why you saw this behavior difference starting from 1.29 before I merge it.

pickledgator, 2022-12-09:
The test I just ran where it resolved the issue was on 1.34 fwiw. I didn’t
explicitly try the in between versions.

FabienDanieau, 2023-01-05:
Hi,

I'm trying to reproduce this example here

> Can someone who's able to reproduce the issue please create a modified version of this script to demonstrate it? https://gist.github.com/jtbandes/9e738c11bb8ec9ba36536837e7990f17

but the cubes are not displayed. I've tried with other primitives as well. Any idea to help me fixing this?
I am using the desktop version 1.35.2 with Debian 12.

[image]

Thanks

jtbandes, 2023-01-05:
See here: https://github.com/foxglove/ws-protocol/issues/278#issuecomment-1324552064