foxglove/studio #4753: Linter for unused npm packages

jtbandes, 2022-11-04:
We should have a lint step that reminds us to remove package dependencies when we stop using them.

Context: https://github.com/foxglove/studio/pull/4450#issuecomment-1301059433

defunctzombie, 2022-11-04:
@jtbandes do you have any experience with a package or tool that you think does this well? I've used some various tools for this with limited viability in the past. Given our complex setup with workspaces I'm hoping there are new tools for this problem?

The first pass on this issue might be finding the available and maintained tools to do this.

defunctzombie, 2022-11-04:
I've moved this into the remove fluentui epic as part of the cleanup efforts to ensure we've done a through job. If we decide to cut this because there's not an appropriate tool we can make that call as part of finishing the epic.

closed 2022-11-15 by defunctzombie