foxglove/studio #4757: How to display multiple mesh models simultaneously

chengchi-qy, 2022-11-04:
**Description**
I'm having a problem when showing the car models(OBJ,dae）.
I am displaying multiple mesh models simultaneously. But when I play it, the positions of the different models blend together.

Right display
[image]

When it starts playing, some of the mesh appears in the wrong place
[image]

Could you tell me some examples for  displaying multiple mesh models simultaneously, thanks.

defunctzombie, 2022-11-04:
Thanks for the bug report. There are many potential causes for what could be happening and the best way we can help resolve this is with a file to reproduce the issue. If you have sensitive data in the file please omit that or create a sample file reproducing the issue.

Other helpful information once we have the file are the frame you are viewing data in and where in the data we can experience the issue.

defunctzombie, 2022-11-15:
Closing this issue for lack of follow-up response. We're happy to take a look but need a better issue description and some sample data that reproduces the issue. If these are provided we can re-open and investigate.

closed 2022-11-15 by defunctzombie