foxglove/studio #4758: Issue rendering ModelPrimitive proto using websocket

pickledgator, 2022-11-04:
**Description**
When sending a SceneUpdate with the ModelPrimitive field populated with binary data from a gltf file, the 3D viewer only renders a straight line instead of the meshes.

- Version:
- OS: 1.28 (since newer versions are blocked by #4732)
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Websocket with protobuf schema

**Steps To Reproduce**
Here's the example packing that I'm using to populate the protobuf:
```
std::vector<uint8_t> ReadFile(const std::string& filepath) {
  std::ifstream instream(filepath, std::ios::in | std::ios::binary);
  std::vector<uint8_t> data((std::istreambuf_iterator<char>(instream)), std::istreambuf_iterator<char>());
  return data;
}
```

```
::foxglove::SceneUpdate scene_update;
auto scene_entity = scene_update.add_entities();
scene_entity->set_frame_id("body");
scene_entity->set_id("model");
scene_entity->set_frame_locked(true);
auto model = scene_entity->add_models();
model->mutable_pose()->mutable_position()->set_x(0.0);
model->mutable_pose()->mutable_position()->set_y(0.0);
model->mutable_pose()->mutable_position()->set_z(0.0);
model->mutable_pose()->mutable_orientation()->set_w(1.0);
model->mutable_pose()->mutable_orientation()->set_x(0.0);
model->mutable_pose()->mutable_orientation()->set_y(0.0);
model->mutable_pose()->mutable_orientation()->set_z(0.0);
model->mutable_scale()->set_x(1.0);
model->mutable_scale()->set_x(1.0);
model->mutable_scale()->set_x(1.0);
auto model_data = ReadFile("seat.gltf");
auto data = model->mutable_data();
data->resize(model_data.size());
std::copy(model_data.begin(), model_data.end(), data->data());
model->set_media_type("model/gltf-binary");
```

Here's a screenshot of how it renders in the 3D panel
[image]

And a screenshot of the raw data view
[image]

jtbandes, 2022-11-04:
Determined in Slack that the issue was `scale` having `y=0` and `z=0`.

closed 2022-11-04 by jtbandes

pickledgator, 2022-11-04:
Thanks for the help here. Nice currently undocumented example for others who might cross this bridge!