foxglove/studio #4759: Add "Open from Data Platform" to Studio Welcome Dialog

2metres, 2022-11-04:


amacneil, 2022-11-07:
What words are we planning to use?

Needs mockup.

esthersweon, 2022-11-08:
My proposal:

[image]

This would link directly to Console's "Recordings" page (or sign in page, if not yet logged in).

Thoughts @amacneil @ocin-rye?

jtbandes, 2022-11-08:
Should we give it some different/brighter styling as a way to upsell more? Also does Data Platform have a logo we can use instead of the "file with globe" icon?

esthersweon, 2022-11-09:
@jtbandes - I prefer the overall option looking similar to the other ones, just so we're not pushing paywalled tools in users' faces. But I'm open to the designers' thoughts. I think a DP-specific icon would be subtle enough though.

Opened PR at https://github.com/foxglove/studio/pull/4785.

amacneil, 2022-11-09:
How about "Browse data"? It's not perfect, but if you are in the mindset that Foxglove hosts data it's sensible.

"Open remote recording" sounds clunky to me, what is a remote recording?

Also, we should link to whatever page in console is the default (currently timeline). If we don't think the default page in console is the best place to browse data, we should fix that separately.

ocin-rye, 2022-11-09:
[image]

Maybe "Browse remote data"? I like "Browse data" and I think "remote" would help clarify that this option is not related to browsing local data.

@esthersweon I like the idea of linking a non logged in users to a sign-in/sign-up page when they select that option. This could be a clear way to communicate the limitations of not signing up for Data Platform.

esthersweon, 2022-11-10:
@amacneil @ocin-rye - We currently use "Open ..." for the other 2 data source options, so I wanted to keep that prefixing consistent. Maybe "Open remote data", if we think "recording" is confusing.

> Also, we should link to whatever page in console is the default (currently timeline). If we don't think the default page in console is the best place to browse data, we should fix that separately.

Why do you feel that this entry point needs to link to the default page for Console? If the CTA is to "Open remote data", I think we should take users to a page where it requires the least amount of work do just that (i.e. 1 click to "View in Studio" from the "Recordings" page). 

The "Timeline" page is a little more removed from that action, so it feels more confusing (more potential for user drop-off). With that said, it still makes sense as the high-level view that a user logging into Console first should see.

amacneil, 2022-11-11:
> The "Timeline" page is a little more removed from that action, so it feels more confusing (more potential for user drop-off). With that said, it still makes sense as the high-level view that a user logging into Console first should see.

This begs the question why the Timeline view is the default in Console, if we expect the main thing people want to do is browse data, and we think the Timeline view is not good for browsing data. Maybe we should change that, but my point is that I don't see why it should be different for Studio users vs people navigating directly to Console.

esthersweon, 2022-11-14:
@amacneil - I'd assume that a user logging directly into Console would more often want a birds' eye view of their data coverage. The app focuses on data management, organization, and tagging, so knowing at a glance what the overall makeup of your imported data looks like would be useful.

On the other hand, someone coming in through Studio is interested in drilling down into specific segments of data – for visualization, analysis, etc. I'd assume that user would be less interested in overall coverage, and more so on worthwhile segments to play back.

amacneil, 2022-11-15:
@esthersweon ok that's fair.

PR #4785 (2022-12-05): Remove "Open file by URL" from Welcome dialog
**User-Facing Changes**
- Move "Open file by URL" in Welcome dialog to under "Open connection" --> "Remote file"
- Replace with callout to load data from Data Platform (open Foxglove console directly)

[image]
[image]

Resolves #4759
commit 3896f7c9 (2022-12-05): Remove "Open file by URL" from Welcome dialog (#4785)

**User-Facing Changes**
- Move "Open file by URL" in Welcome dialog to under "Open connection"
--> "Remote file"
- Replace with callout to load data from Data Platform (open Foxglove
console directly)

<img width="777" alt="Screen Shot 2022-11-15 at 1 11 51 PM"
src="https://user-images.githubusercontent.com/6993359/202026370-da7a7617-158c-46e2-bf98-84a991e94d30.png">
<img width="781" alt="Screen Shot 2022-11-15 at 1 11 43 PM"
src="https://user-images.githubusercontent.com/6993359/202026372-9696862e-de35-4bcd-97f9-fa1a3cf67ee9.png">

Resolv [... cut]
closed 2022-12-05 by esthersweon