foxglove/studio #4797: /clock topic is ignored for ws-protocol data source

jhurliman, 2022-11-10:
Other live data sources look for a topic named `/clock` and use that to set the data source time. The ws-protocol player does not appear to do this, causing visualization problems for looping bag playback.

jtbandes, 2022-11-10:
Since the foxglove websocket protocol is not ROS-specific, there's no reason it should look for a topic named `/clock`. If this is necessary then we should probably update the protocol to add some kind of time op.

On the other hand, the ws player does use the receiveTime to set its currentTime:

https://github.com/foxglove/studio/blob/48c21f74839bc04c20ea1909948ae887f45e2a25/packages/studio-base/src/players/FoxgloveWebSocketPlayer/index.ts#L261

jhurliman, 2022-11-10:
Just `receiveTime` is insufficient. If I have low-frequency markers and transforms that I want to interpolate at 100hz using the `--clock 100` argument to `ros2 bag play`, then setting `receiveTime` doesn't have the right effect (because nothing would be subscribing to the high-frequency `/clock` topic).

I can add this in once foxglove_bridge is communicating the "ros1" / "ros2" profile in `serverInfo`.

jtbandes, 2022-11-10:
If you are interpolating at 100hz if no new data has been received, isn't that extrapolation? I thought we didn't support TF extrapolation.

jhurliman, 2022-11-10:
We support TF interpolation, and you can publish TFs with header.stamp > currentTime. This is generally used for motion prediction / motion planning.

jhurliman, 2022-11-11:
Closing in favor of https://github.com/foxglove/studio/issues/4805

closed 2022-11-11 by jhurliman