foxglove/studio #4801: Plot panel memory usage rapidly increases over time, leading to long GC cycles for dense layouts

DavidXenakis, 2022-11-10:
**Description**
We have been experiencing poor performance and crashes related to runaway memory use. One of the culprits, we believe, is the plot panel. For particularly dense layouts including many plots and series, chrome's performance profile shows memory usage rapidly increasing during playback, even after all data has been loaded and cached. 

Sawtooth memory usage pattern.
[image]

Please investigate whether or not the plot panel's memory footprint can be reduced.

- Version:
- OS: All
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Any? It repros from the studio.foxglove.dev sample dataset.
- ROS distro (if applicable): n/a

**Steps To Reproduce**
1. Visit https://studio.foxglove.dev/?ds=sample-nuscenes&layoutId=0daa2893-d4b6-4146-a4b7-9285c9637358&time=2018-07-24T03%3A28%3A49.082513669Z
2. Load all the data by playing through the log.
3. Run a performance profile during playback to capture memory usage.

**Expected Behavior**

**Actual Behavior**

defunctzombie, 2022-11-10:
Layout:

<details>
<pre>
{
  "configById": {
    "Plot!39p1t8k": {
      "title": "Plot",
      "paths": [
        {
          "value": "/gps.latitude",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/gps.longitude",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/gps.altitude",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/gps.position_covariance_type",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/drivable_area.pose.position.x",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/drivable_area.pose.position.y",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/drivable_area.pose.position.z",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/drivable_area.pose.orientation.x",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/drivable_area.pose.orientation.y",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/drivable_area.pose.orientation.z",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/drivable_area.pose.orientation.w",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
          "value": "/drivable_area.column_count",
          "enabled": true,
          "timestampMethod": "receiveTime"
        },
        {
[... 1214 lines cut ...]
      "second": {
        "first": "Plot!n93ptq",
        "second": "Plot!2ojg7qw",
        "direction": "column"
      },
      "direction": "column"
    },
    "second": {
      "first": {
        "first": "Plot!19ldvau",
        "second": "Plot!2s45z45",
        "direction": "column"
      },
      "second": {
        "first": "Plot!2pejgzc",
        "second": "Plot!370cv4i",
        "direction": "column"
      },
      "direction": "column"
    },
    "direction": "column"
  }
}
</pre>
</details>

foxymiles, 2022-11-14:
Thanks for the detailed report @DavidXenakis . What version of Chrome and what OS are you using to reproduce this?

DavidXenakis, 2022-11-14:
This was reproducible for me on Chrome version 107.0.5304.87 on MacOS Monterey.

defunctzombie, 2022-11-15:
[image]

I was able to reproduce this result on my mac. The drop in memory occurs when I let playback loop from the start. What I am seeing here looks like memory buildup during playback and release when playback restarts.

jtbandes, 2022-11-23:
This should help a lot once the data has been fully buffered: https://github.com/foxglove/studio/pull/4884

jtbandes, 2022-12-01:
This will help reduce extra renders when seeking: https://github.com/foxglove/studio/pull/4885

defunctzombie, 2022-12-13:
@jtbandes is there further work planned on this or are we waiting for feedback? Do we believe the original issues described are resolved?

jtbandes, 2022-12-13:
We are waiting for feedback.

jtbandes, 2022-12-21:
Got confirmation from @DavidXenakis that memory growth is greatly reduced. Closing out this issue, and any further issues can be tracked separately.

closed 2022-12-21 by jtbandes