foxglove/studio #4808: I'm unable to render the visual collision scene assets(format STL) on the 3D panel, similar to what I would see on Rviz

kevin-kindred, 2022-11-14:
**Description**
I'm unable to render the visual collision scene assets(format STL) on the 3D panel, similar to what I would see on Rviz

- Version: 1.31.0
- OS: MacOS 12.6.1
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): rosbridge
- ROS distro (if applicable): ROS2 Humble

**Steps To Reproduce**
1. Start a ROS2 application that publishes assets & robot description on the topic /robot_description(This is for a serial manipulator and using MoveIt)
2. Install ros-humble-rosbridge-suite & start rosbridge-server using `ros2 launch rosbridge_server rosbridge_websocket_launch.xml`
3. Open connection to robot using rosbridge on native client/browser
4. Create a 3D panel

**Expected Behavior**
I can see the serial manipulator similar to how it's seen in Rviz
[image]

**Actual Behavior**
I see just the links & their reference frames
[image]

I also see this error on the 3D panel settings
[image]

jtbandes, 2022-11-14:
> Open connection to robot using rosbridge on native client/browser

Please note that package:// urls only work in the Desktop app, as documented here: https://foxglove.dev/docs/studio/panels/3d#add-unified-robot-description-format-urdf

Additionally, you may need to configure the ROS_PACKAGE_PATH here:
[image]

jtbandes, 2022-11-18:
@kevin-kindred Any update on this? Were you able to get it working properly?

jtbandes, 2022-12-08:
I'm going to close out this issue since we haven't heard back in a few weeks. Let us know if you are still having trouble and we can continue helping out.

closed 2022-12-08 by jtbandes

kevin-kindred, 2023-02-13:
Apologies for not getting back earlier. I believe `ROS_PACKAGE_PATH` is not a variable that ROS2 supports anymore https://github.com/ros/ros_environment/issues/25

Also for a bit more context, I'm trying to visualize the manipulator from UR using their ROS2 description https://github.com/UniversalRobots/Universal_Robots_ROS2_Description