foxglove/studio #4809: Setting to Automatically Subscribe to Visualization Topics in 3D Tab

cody-sheff, 2022-11-14:
Small feature request to add some sort of setting or toggle that will automatically subscribe to all detected topics when the 3D tab is open. As-is, even with the ability to show/hide all it's a bit tedious to set things up when there are a large number of changing geometries.

jtbandes, 2022-11-14:
Are you saying that your set of topics often changes dynamically? (Which data source are you using?) Or just that you would always like a new panel to start with subscriptions to all topics?

cody-sheff, 2022-11-14:
We run tests that different topics, and having a way to automatically subscribe to all would speed things up considerably. 

The main use-case for our application is using this over websocket using the ws-foxglove server. The only way to confirm that a channel has been properly set-up before sending messages is to see if it is subscribed to so, in order to make sure that critical messages such as geometries are properly sent, we wait for all those topics to be subscribed to before continuing execution. While having profiles with the proper topics subscribed to speeds this up, an auto-subscribe functionality would make this much easier.

jtbandes, 2022-11-14:
Related: 

- https://github.com/foxglove/studio/issues/3643
- https://github.com/foxglove/studio/issues/3692