foxglove/studio #4822: QOL improvements for mobile devices

AndBondStyle, 2022-11-15:
We're using web version of foxglove on an android tablet as a remote controller for our robot. Surprisingly, it worked pretty well out of the box. However, there're some major inconveniences we've faced so far:

### 1. Custom extensions (.foxe) can only be installed via drag-and-drop

As you can imagine, this is completely impossible to do on a mobile device. I would suggest adding a "load extension" button to extensions panel, which just opens a normal file dialog.

### 2. Unable to load layouts from file

You use `window.showOpenFilePicker` to pick a layout file, but this function is not implemented on mobile. Maybe use a shim?

### 3. Unable to use "publish point" in a 3D panel

Probably, some modifications should be done to `ThreeDeeRender/Input.ts` to support touch events. I see some callbacks, but currently they are empty or don't do anything useful. Below is my patch for this file, it's a bit messy, but seems to be working fine. Another idea is to use [pointer events](https://developer.mozilla.org/en-US/docs/Web/API/Pointer_events), which should work properly both on desktop and mobile devices.

```diff
+  private convertEvent = (event: TouchEvent): MouseEvent | null => {
+    // Helper function to convert TouchEvent to fake MouseEvent
+    let touch = event.touches[0];
+    if (!touch || !event.target) return null;
+    let rect = (event.target as Element).getBoundingClientRect();
+    // Other code only uses .offsetX/Y, so other attributes are not necessary
+    let ev = { offsetX: touch.pageX - rect.left, offsetY: touch.pageY - rect.top };
+    return ev as MouseEvent;
+  }
+
   private onTouchStart = (event: TouchEvent): void => {
-    const touch = event.touches[0];
-    if (touch) {
-      this.startClientPos = new THREE.Vector2(touch.clientX, touch.clientY);
-    }
     event.preventDefault();
+    let ev = this.convertEvent(event);
+    if (!ev) return;
+    this.startClientPos = new THREE.Vector2(ev.offsetX, ev.offsetY);
+    this.updateCursorCoords(ev);
   };
 
   private onTouchEnd = (event: TouchEvent): void => {
     event.preventDefault();
+    // "touchend" doesn't have any touches, so we can't create a proper event
+    let ev = {} as MouseEvent;
+    this.emit("click", this.cursorCoords, this.worldSpaceCursorCoords, ev);
   };
```

### 4. Auto focus on inputs frequently causes keyboard to pop up

This happens every time you want to quickly adjust settings for some panel. Minor inconvenience, but very annoying. I would suggest to disable auto focus on mobile devices completely.

### 5. Marketplace extensions are disabled in web version

This is strange, because installing custom extensions worked okay so far (with some dirty fix for problem # 1). Is there a reason for this decision?

defunctzombie, 2022-11-15:
> ### 1. Custom extensions (.foxe) can only be installed via drag-and-drop
> As you can imagine, this is completely impossible to do on a mobile device. I would suggest adding a "load extension" button to extensions panel, which just opens a normal file dialog.

When you sign into your foxglove account we can provision extensions to your account from your organization. That is the approach we have in place for distributing your extension to your organization and web app where you cannot drag-drop the extension.

AndBondStyle, 2022-11-15:
> When you sign into your foxglove account we can provision extensions to your account from your organization. That is the approach we have in place for distributing your extension to your organization and web app where you cannot drag-drop the extension.

We rely on dockerized environments that are completely ready to go without additional interaction, signing into external accounts etc. Also, I don't fully understand how to do local testing of extensions without pushing in to organization. Do I really need to do reupload for every build of extension?

defunctzombie, 2022-11-15:
> Do I really need to do reupload for every build of extension?

We recommend the desktop app for testing the extension during development. There is a `local-install` command for the extension to install locally and test with the desktop app.

> It's also a bit silly to use cloud service

This is our current choice for managing extensions within an organization.