foxglove/studio #4824: Memory leak when switching layouts containing 3D panel

asyncanup, 2022-11-15:
Seeing memory leaks when switching back and forth between layouts containing 3D panel, and when unmounting/remounting the app inside a host app.

The root cause for the memory leak is likely the following: the new 3D panel is implemented as an extension, and there does not appear to be any mechanism implemented in FG Studio to clean up after extension panels, or the way the 3D panel is wrapped as extension does not detect unmounting of the extension wrapper. Almost all objects that linger around as leak are allocated in the 3D panel.

We observed that the 3D panel gets initialized actually twice during a single bringup (in live mode), but never cleared. And I am still measuring about 10-20MB leak per cycle (of unmounting and remounting the app)

- Version: v1.31.0
- OS:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
- ROS distro (if applicable):

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

sarbjit-longia, 2022-11-17:
I have worked on reproducing the issue with app mount/unmount and was able to reproduce the issue. I do understand that this might not be the normal use case for foxglove and might never happen with foxglove studio but its a big show stopper for us so would appreciate if we can spend time on this and fix it.

I have created a document that can be used to follow the steps in order to reproduce the issue. Document contains following:

- Steps to reproduce
- Drive session data data used
- Layout used for reproducing
- Video of the steps I used along with final Ah-Snap happening.
- Heap snapshots that I used to get some data.
- Code analysis for what I found out.

Please keep in mind that the code analysis is done to the best of my knowledge and is possible that I am looking in wrong direction so please use your own judgement and knowledge to fix this issue. 

**Steps:**

- Started with foxglove latest code from github.
- Added the code to add On-Off button([Patch](https://drive.google.com/file/d/1jomPy7No6bewfD_Uern-PSFhhA2pU2mv/view?usp=share_link)).
- Used data source from [link.](https://drive.google.com/file/d/1L9rmnlRcAGQIKwdcTK16YUMMm-NtX-Vt/view?usp=share_link)
- Used [layout](https://drive.google.com/file/d/1MEGE5ARAWT5yKbHbzxkMmaSuiRaS8vZT/view?usp=share_link).
- Used on-off button few times

[Google Doc](https://docs.google.com/document/d/1X9g6VuHUyoquzfQ3NX9pqgYYCmzbPqvuXo-xeL6arWI/edit?usp=sharing)

snosenzo, 2022-11-17:
Thanks so much for digging into this! Will start looking into it today, will let you know what I find

snosenzo, 2022-11-17:
These are my findings after doing some digging. 

These memory leaks  are primarily created and can be tested by loading a new source after loading the initial source without reloading the page and checking to see that the JS heap is larger than after initially buffering the initial source. This got me the same results as your mounting/unmounting patch.
 
Also I was able to test a version where the renderer's `.dispose()` function was called and unmounted and that didn't seem to have an impact on the memory leak, which lead me to the image panel. 

From this I was able to narrow the scope of the largest part of the memory leak to the ImageAnnotation topics by testing them being read in by the ImagePanel and RawMessage panel, resulting in ~50MB heap allocations per buffered topic every time a new source is selected. Looking into finding the source of this, but I did have a few questions. 

What is your current flow that's leading you to run into these leaks? (what panels are being used and types of topics are being read) 
What is an acceptable level of memory leaks when going through this flow?

jtbandes, 2022-11-18:
Possibly related: https://github.com/foxglove/studio/pull/2904

sarbjit-longia, 2022-11-18:
Thanks thats promising that you were able to narrow down it to Image panel. 

We are using this app embedded in another umbrella app which has a bunch of apps and when user click on some other app it unloads foxglove UI and loads that app. In our use case which is a show stopper for us we use a 3D panel and an Image panel which is rendering the video from topic. 

The ideal value we are looking for is 0 memory leak but from your question it seems that is challenging. What is the best you can achieve?

sarbjit-longia, 2022-11-18:
> * Possibly related: [Fix memory leak of player state from previous players #2904](https://github.com/foxglove/studio/pull/2904)

That fix was already there in my testing. I think it might be similar in nature but different memory leak. I did attached the document with screenshots and video that can help you isolate this.

snosenzo, 2022-11-29:
Just wanted to give an update to say that I'm still working on this! I believe I've closed out a number of leaks that are keeping the player around after a new source is loaded and am now addressing the remaining leaks that are keeping the player in memory in the 3D panel (which is indeed a part of the issue).

asyncanup, 2022-11-29:
Thanks for the update. Appreciate the persistent effort.

snosenzo, 2022-12-08:
So as of this PR (https://github.com/foxglove/studio/pull/4946) getting merged, this should be much less of a problem across all layouts regardless of whether they contain a 3D or extension panel. 

A few things contributed to these leaks:
Primarily panel extensions not gracefully unmounting and dereferencing themselves from the window. This caused the renderer to not properly dispose of the GL contexts, geometry, textures, and listeners. This compounded with them holding reference to old `IterablePlayers` storing buffered data.
These were addressed in the following PRs:
https://github.com/foxglove/studio/pull/4914
https://github.com/foxglove/studio/pull/4922

The other issued stemmed from other parts of the application holding onto old players with buffered data via closure contexts and detached dom elements refs. 
This was addressed in the following PRs:
https://github.com/foxglove/studio/pull/4907
https://github.com/foxglove/studio/pull/4906
https://github.com/foxglove/studio/pull/4924
https://github.com/foxglove/studio/pull/4943

There are other instances of old `IterablePlayers` being kept in memory that are a little harder to track down. To mitigate these I put up a PR to clear the buffers in the old players so that they don't take up nearly as much space and the memory leak due to it will be more negligible. Addressed here: https://github.com/foxglove/studio/pull/4906

Thanks for the patience and hopefully this is a sufficient enough improvement for your usecase. Do let me know if further improvements are needed

snosenzo, 2023-01-09:
after talking with @sarbjit-longia on slack it sounds like the issue has been fixed and looks good now -- closing

closed 2023-01-09 by snosenzo