foxglove/studio #4828: 🔪 Nested enum annotations

amacneil, 2022-11-16:
## Chapter 1

ROS messages do not have an enum types. Informally, they solve this using constants (e.g. [DiagnosticStatus](https://docs.ros.org/en/noetic/api/diagnostic_msgs/html/msg/DiagnosticStatus.html)):

```
byte OK=0
byte WARN=1
byte ERROR=2
byte STALE=3

byte level # level of operation enumerated above 
```

However, there is no strict mapping from the constants to the `level` field. This leads to predictable chaos when a message contains multiple enums (e.g. [Marker](http://docs.ros.org/en/noetic/api/visualization_msgs/html/msg/Marker.html)):
```
uint8 ARROW=0
uint8 CUBE=1
uint8 SPHERE=2
uint8 CYLINDER=3
uint8 LINE_STRIP=4
uint8 LINE_LIST=5
uint8 CUBE_LIST=6
uint8 SPHERE_LIST=7
uint8 POINTS=8
uint8 TEXT_VIEW_FACING=9
uint8 MESH_RESOURCE=10
uint8 TRIANGLE_LIST=11

uint8 ADD=0
uint8 MODIFY=0
uint8 DELETE=2
uint8 DELETEALL=3

int32 type
int32 action
```

## Chapter 2

Foxglove Studio, not content with this existing level of chaos, also supports taking things to the [next level](https://foxglove.dev/docs/studio/annotating-data#after-annotating):
test_msgs/Test
```
test_msgs/Color my_color__foxglove_enum
uint8 my_color
```

test_msgs/Color
```
uint8 RED=1
uint8 YELLOW=2
uint8 BLUE=3
```

What is happening here? The `test_msgs/Test` message contains a `my_color` enum, but also has a field named `my_color__foxglove_enum` of type `test_msgs/Color`. The `test_msgs/Color` doesn't contain any actual fields, so `my_color__foxglove_enum` is always going to be empty.

I guess Foxglove Studio then has some logic to detect based on the field naming that `my_color` is actually an enum as defined by `test_msgs/Color`. This can't be a simple comment because then ROS would not bundle the `test_msgs/Color` in bag files. But adding a real field with a special name seems super convoluted.

[image]

@jtbandes also discovered that this does not work in ROS 2 because field names cannot contain double underscore.

## Conclusion

The functionality described in Chapter 1 is reasonable (best effort based on established ROS conventions), but can we remove the functionality described in Chapter 2? My hunch is that this is an internal Cruiseism that was never removed. 

cc @defunctzombie

defunctzombie, 2022-11-16:
I think this is a use of the .MSG syntax that does not exist in the wild outside of Cruise. I would vote we remove this from the docs and schedule this feature for removal from Studio.

I've never seen any guide anywhere in the ROS ecosystem that would encourage folks to use _empty_ message files in this manner and I find the additional "field" being present confusing when considered with the typical layout of a .MSG file. I've only seen the opposite - where you are encouraged to write your constants in the same .MSG file and prefix them with the field name.

jtbandes, 2022-11-16:
> However, there is no strict mapping from the constants to the level field. This leads to predictable chaos when a message contains multiple enums (e.g. [Marker](http://docs.ros.org/en/noetic/api/visualization_msgs/html/msg/Marker.html))

Actually, there is no inherent problem with multiple blocks of constants, as long as their names don't overlap. The app does support this (see below). The constants just have to be located next to the appropriate field, and their type has to match the field type, both of which are not true in the Marker definition.

https://github.com/foxglove/studio/blob/e7cad8306591e6b86e8fe12f1969152860463657/packages/studio-base/src/util/selectors.test.ts#L72-L102

> I think this is a use of the .MSG syntax that does not exist in the wild outside of Cruise.

Just to be clear, this is speculation and we don't have any data on this whatsoever. The feature is documented so it is very possible that some people are using it — but probably not a _lot_ of people.

>  I've only seen the opposite - where you are encouraged to write your constants in the same .MSG file and prefix them with the field name

The reason this feature came about is that writing constants in the .msg file doesn't handle:
- using the same name for two enums (e.g. if they both have an UNKNOWN)
- using the same enum for two fields in the message or in two separate messages – in C++ land you want to use `tracked_object.type = ObjectType::PEDESTRIAN` and so it's helpful to have an ObjectType msg that contains the constants so it can be reused across tracked_object messages, predicted_object messages, etc. I believe this pattern was already in use at Cruise, but we determined that the constant definitions didn't get written into the bag, so we came up with the technique of including an empty annotation field which forces it to include the definitions (and also tells webviz/studio which enum to use for the field).

I'm not going to argue strongly for keeping this feature but trying to provide context on why it exists. The true solution is for ROS to add support for enums, but that's clearly not going to happen any time soon (https://github.com/ros2/rosidl/issues/260). And even if it did that would be in ROS 2, whereas we've determined this annotation feature only works for ROS 1.

- See also https://github.com/foxglove/studio/issues/2214

amacneil, 2022-11-17:
Anyone want to add a metric for this feature so we can check usage before killing it?

defunctzombie, 2022-11-17:
> Anyone want to add a metric for this feature so we can check usage before killing it?

What type of usage would result in us keeping it? 1 user? 10?