foxglove/studio #4846: Link 3D Panels

banisadr, 2022-11-17:
Ability to sync or "lock" multiple 3D panels together in a layout such that pan/zoom in one does the same in the other linked panels. Needs both the ability to do 1-time sync of panels as well as ability to lock panels together.

foxymiles, 2022-11-21:
Do we want to sync the complete camera state or just pan zoom? If we only sync pan and zoom that might not make sense for different views with different initial camera states right?

foxymiles, 2022-11-21:
Also if there is a mix of 3d and 2d views in a layout how should we sync them?

foxymiles, 2022-11-21:
Screenshot of the feature in the old 3d panel:

[image]

jtbandes, 2022-11-22:
Assigning to @snosenzo to get #4859 over the finish line with the changes requested in https://github.com/foxglove/studio/pull/4859#issuecomment-1322801204

closed 2022-12-04 by foxymiles