foxglove/studio #4852: Image and View Frustum seem overly stretched in 3D view

danielgordon10, 2022-11-18:
**Description**
When using the CameraCalbiration Proto and a RawImage proto to draw an image in a 3D view, the images look stretched. Square images seem to stay square, but non-square get warped in some odd way.

**Steps To Reproduce**

Run the following code
```python
# Install deps:
#   pip3 install foxglove-websocket foxglove-schemas-protobuf
# Run:
#   python3 test_img.py

import math
import asyncio
import time
from base64 import b64encode
from typing import Set, Type
from foxglove_websocket import run_cancellable
from foxglove_websocket.server import FoxgloveServer, FoxgloveServerListener
from foxglove_websocket.types import ChannelId
from foxglove_schemas_protobuf.RawImage_pb2 import RawImage
from foxglove_schemas_protobuf.CameraCalibration_pb2 import CameraCalibration
import google.protobuf.message
from google.protobuf.descriptor_pb2 import FileDescriptorSet
from google.protobuf.descriptor import FileDescriptor

def build_file_descriptor_set(
    message_class: Type[google.protobuf.message.Message],
) -> FileDescriptorSet:
    """
    Build a FileDescriptorSet representing the message class and its dependencies.
    """
    file_descriptor_set = FileDescriptorSet()
    seen_dependencies: Set[str] = set()

    def append_file_descriptor(file_descriptor: FileDescriptor):
        for dep in file_descriptor.dependencies:
            if dep.name not in seen_dependencies:
                seen_dependencies.add(dep.name)
                append_file_descriptor(dep)
        file_descriptor.CopyToProto(file_descriptor_set.file.add())  # type: ignore

    append_file_descriptor(message_class.DESCRIPTOR.file)
    return file_descriptor_set

async def main():
    class Listener(FoxgloveServerListener):
        def on_subscribe(self, server: FoxgloveServer, channel_id: ChannelId):
            print("First client subscribed to", channel_id)

        def on_unsubscribe(self, server: FoxgloveServer, channel_id: ChannelId):
            print("Last client unsubscribed from", channel_id)

    async with FoxgloveServer("0.0.0.0", 8765, "example server") as server:
        server.set_listener(Listener())
        img_chan_id = await server.add_channel(
            {
                "topic": "example_msg",
                "encoding": "protobuf",
                "schemaName": RawImage.DESCRIPTOR.full_name,
                "schema": b64encode(
                    build_file_descriptor_set(RawImage).SerializeToString()
                ).decode("ascii"),
            }
        )
        calib_chan_id = await server.add_channel(
            {
                "topic": "example_calib",
                "encoding": "protobuf",
                "schemaName": CameraCalibration.DESCRIPTOR.full_name,
                "schema": b64encode(
                    build_file_descriptor_set(CameraCalibration).SerializeToString()
                ).decode("ascii"),
            }
        )

        i = 0
        while True:
            i += 1
            await asyncio.sleep(0.1)
            now = time.time_ns()

            img = RawImage()
            img.timestamp.FromNanoseconds(now)
            img.frame_id = "img"
            img.width = 100
            img.height = 100
            img.encoding = "bgra8"
            data = bytearray()
            for y in range(img.height):
                for x in range(img.width):
                    data.extend([int(y/img.height * 255), int(x/img.width * 255), 127, 255])
            img.data = bytes(data)
            await server.send_message(img_chan_id, now, img.SerializeToString())

            calib = CameraCalibration()
            calib.timestamp.FromNanoseconds(now)
            calib.frame_id = "img"
            calib.width = img.width
            calib.height = img.height
            calib.distortion_model = "plumb_bob"

            fx = 30
            fy = 30
            cx = img.width / 2
            cy = img.height / 2
            tx = 0
            ty = 0
            calib.K[:] = [fx, 0, cx, 0, fy, cy, 0, 0, 1]
            calib.P[:] = [fx, 0, cx, tx, 0, fy, cy, ty, 0, 0, 1, 0]

            await server.send_message(calib_chan_id, now, calib.SerializeToString())

if __name__ == "__main__":
    run_cancellable(main())
```
And connect to the websocket 

**Actual Behavior**
A width height ratio of 1:1 results in square. This is expected.
[image]

Notice that with a ratio of sqrt(2):1 the side lengths appear 2:1
[image]

With a width height ratio of 2:1 the side lengths look like 4:1
[image]

The squashing is evident in real data as well. In the Foxglove samples, I saw the same behavior. Look how the W is stretched as compared with the actual image.
[image]

danielgordon10, 2022-11-18:
I suspect something about https://github.com/foxglove/studio/blob/bebfa77c2b7a6ad810ed8b97354be21546d41ecc/packages/den/image/PinholeCameraModel.ts#L168 may be wonky. I'm not sure you want to be multiplying by the width here. But I don't fully understand this code, so I could be wrong.

jhurliman, 2022-12-09:
@danielgordon10 take a look at <https://github.com/foxglove/studio/pull/4953> and let me know if things look correct on your end.

closed 2022-12-09 by jhurliman

danielgordon10, 2022-12-13:
Looks fixed to me. Thanks.

PR #4953 (2022-12-09): [3D] Fix projected image dimensions, configurable planar projection factor
**User-Facing Changes**
- [3D] Fixed aspect ratio for camera frustums and images
- [3D] Configurable "Planar Projection Factor" for camera frustums and images

**Description**
This PR makes a few changes to the math in PinholeCameraModel to make the results match the ROS PinholeCameraModel as closely as possible. Tests have been added with ground truth from ROS Noetic. Rectification correctly normalizes by focal length instead of image dimensions. Five Jacobi iterations are used instead of three, with a bypass if the di [... cut]
commit a093e9f2 (2022-12-09): [3D] Fix projected image dimensions, configurable planar projection factor (#4953)

**User-Facing Changes**
- [3D] Fixed aspect ratio for camera frustums and images
- [3D] Configurable "Planar Projection Factor" for camera frustums and
images

**Description**
This PR makes a few changes to the math in PinholeCameraModel to make
the results match the ROS PinholeCameraModel as closely as possible.
Tests have been added with ground truth from ROS Noetic. Rectification
correctly normalizes by focal length instead of image dimensions. Five
Jacobi iterations are used instead of three, with a bypass  [... cut]