foxglove/studio #4868: cant load file from nfs

chasingw, 2022-11-22:
I mount file by nfs in ubuntu 18.04 use command like `sudo mount -t nfs xxxx:xxx xxxx`. I can use the file in other software like files, brower. but I cant see the file in foxglove.

jtbandes, 2022-11-22:
Do you get any errors in the app?

chasingw, 2022-11-23:
> Do you get any errors in the app?

no errors. when I drop a file to the studio, it display `drop a file here` and nothing happened. if I click Open local file and I cant see the nfs mounted directory in the file dialog

jtbandes, 2022-11-23:
Any errors in the developer console?

chasingw, 2022-11-23:
> Any errors in the developer console?

sorry, I use the desktop app in ubuntu 18.04, how to show the developer console

jtbandes, 2022-11-23:
Alt > View > Advanced