foxglove/studio #4871: Expanding a node in Raw Messages expands the whole tree

bclinkinbeard, 2022-11-22:
**Description**
After collapsing a large message, opening a single field in a Raw Messages topic expands the whole tree.

- Version:
- OS: All
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Sample data on studio.foxglove.dev
- ROS distro (if applicable):

**Steps To Reproduce**
1. Open https://studio.foxglove.dev/?ds=sample-nuscenes&layoutId=b32d8240-c32c-4da3-85fa-0f9d9f3afa7a&time=2018-07-24T03%3A28%3A47.703844000Z
2. Use the button at top left to collapse
[image]
3. Click to expand any field, like timestamp in the image above

**Expected Behavior**
The timestamp field is expanded but the rest of the object remains collapsed.

**Actual Behavior**
The whole object is expanded.
[image]

amacneil, 2022-11-23:
Only seems to happen with a nested message message path, not a top level message:

https://user-images.githubusercontent.com/637671/203449365-0c0cf7ea-083a-4f51-adcd-33f8aa513bfc.mov

amacneil, 2022-12-08:
@bclinkinbeard after a long discussion, it sounds like this is working as designed, due to complexity of nested messages keeping track of expanded/collapsed state while messages play.

However it is far from ideal so going to keep this ticket open, but downgrading it from a bug.