foxglove/studio #4873: Plot panel does not load Data Platform data sometimes

amacneil, 2022-11-22:
**Description**
Sometimes when I click through from Console to Studio, the plot panel does not load. Adding/removing a topic fixes it.

https://user-images.githubusercontent.com/637671/203414234-297c31a7-3018-45df-846e-f90baf2328ba.mov

Happens ~50% of the time. I have not been able to reproduce it with _only_ a plot panel, although maybe that is coincidence.

Hitting Cmd+R seems like it does not trigger this issue, it almost always seems to happen when I click through from Console.

- Version: Web
- OS: Mac
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Data Platform
- ROS distro (if applicable): N/A

amacneil, 2022-11-30:
Can't repro, will monitor and reopen if it happens again.

jtbandes, 2023-01-03:
We think this was fixed with https://github.com/foxglove/studio/pull/5047

closed 2023-01-03 by jtbandes