foxglove/studio #4892: Data are accumulated on the source frame instead of the display frame when decay time > 0

cedrusx, 2022-11-28:
**Description**
In the 3D panel, the accumulation of laser scan (not sure about other types) seems to be problematic when setting a positive value of decay time. I expect its points to be accumulated on the display frame, like in RViz. But it seems to be accumulating on the data source frame or other way. 

- Version: 1.32.0-dev
- OS: Windows 11
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
- ROS distro (if applicable):

**Steps To Reproduce**

Open a bag that has a laser scan topic, where the scan data are associated on a sensor frame (e.g. "laser"), and the sensor frame is moving w.r.t. a world frame (e.g. "world"), with correct transformations on /tf.
Open a 3D panel, in its settings, set the Display frame to be the world frame and the Follow mode to be Fixed. Then turn on the scan topic, and set its decay time to a positive number such as 100.

**Expected Behavior**
The scan points are accumulated on the world frame, so that they form the shape of the environment when the robot is moving.

**Actual Behavior**
The accumulated points move with the robot.

See an example below: 
(on the right hand side there is the accumulation of the same data in RViz, for comparison)

[image]

jtbandes, 2022-12-08:
Hi @cedrusx, would you be able to provide a sample bag we can use to test the issue?

This is probably related to the fact that we treat point clouds & laser scans as "frame_locked=false", meaning their locations are set in the fixed frame when they are received.

jtbandes, 2022-12-15:
Another simpler sample file that can be used to reproduce the issue: [gh4892_decay_time.mcap.zip](https://github.com/foxglove/studio/files/10233026/gh4892_decay_time.mcap.zip)

jtbandes, 2022-12-15:
Thanks for the report. I've fixed some issues in #4986 and I believe it's now behaving correctly! Here's what your sample data looks like now:

https://user-images.githubusercontent.com/14237/207955675-0ffb7406-3bff-4a8e-8db9-2ffaca5133f8.mov

closed 2022-12-15 by jtbandes

cedrusx, 2022-12-18:
Awesome! Thank you for the quick fix!

PR #4986 (2022-12-15): Fix decay issues with PointClouds and LaserScans
**User-Facing Changes**
Fixed issues with PointCloud and LaserScan messages displaying incorrectly in the 3D panel when their `pose` changed, or when the Decay Time setting was used.

**Description**
Fixes #4892
Fixes FG-967

Issues fixed:
- The `pose` of the latest history item was not being updated after it was initially created.
- LaserScans needs `itemSize=1` for the `position` attribute, but it was sometimes using the shared `createGeometry` function which uses `itemSize=3`

Test data: [gh4892_decay_time.mcap.zip](https://github.com/foxglov [... cut]
commit 04aa8b22 (2022-12-15): Fix decay issues with PointClouds and LaserScans (#4986)

**User-Facing Changes**
Fixed issues with PointCloud and LaserScan messages displaying
incorrectly in the 3D panel when their `pose` changed, or when the Decay
Time setting was used.

**Description**
Fixes #4892
Fixes FG-967

Issues fixed:
- The `pose` of the latest history item was not being updated after it
was initially created.
- LaserScans needs `itemSize=1` for the `position` attribute, but it was
sometimes using the shared `createGeometry` function which uses
`itemSize=3`

Test data:
[gh4892_decay_time.mcap.zip](https://github.co [... cut]