foxglove/studio #4893: Standardize drag-and-drop pattern across panels

tharejas, 2022-11-28:
Per discussion on [slack](https://nvidia.slack.com/archives/C026X05N8CU/p1669665354337969?thread_ts=1669664616.601369&cid=C026X05N8CU), panels with custom toolbars (plot, raw message, diagnostics, log etc) have a restricted drag area and the user needs to click on the gear (settings) icon to drag these panels. This is unintuitive and confusing, pls improve.

- Version: Foxglove Studio version 1.32.0-dev

**Steps To Reproduce**
Try to drag a plot panel by clicking on the toolbar

**Current Experience**
No "hand" icon indicating its draggable, cannot drag

**Desired Experience**
All panels draggable by clicking/dragging their toolbar

amacneil, 2022-11-29:
I don't think this is solvable until we remove all custom controls from panel toolbars.

jtbandes, 2022-11-29:
Related:
- https://github.com/foxglove/studio/issues/4880
- https://github.com/foxglove/studio/issues/4881
- https://github.com/foxglove/studio/issues/4888