foxglove/studio #4894: Support for urdf.xacro file with namespaces to visualise robot_description in 3D panel

Dave-van-der-Meer, 2022-11-29:
In our lab, we have multiple robots of the same model, and with the switch to ROS2, we set them up to have proper namespaces for each robot. Since we have several Leo Rovers, we gave them the namespace "leo01", "leo02", "leo03", ...

The namespace is applied to the node names, topic names, service names, and URDF links and joints.

Currently, we can load the generic robot_description model into the 3D panel using the urdf.xacro file, but it does not apply a namespace to the robot model. as a result, we have the TF for `/leo01/base_link` and `/leo02/base_link`, but the 3D model will be called `/base_link` without the namespace.

The namespace is loaded when calling the robot_description, and it is not hard coded into the URDF file. This way, we can reuse it for any new robot and also update it for all the robots at once. When we actually convert the urdf.xacro file into a pure URDF file and apply the namespace by hard-coding them into the URDF file, it can properly assign the namespace and the 3D model matches with the published TF.

My feature request would be to add support for namespaces in the urdf.xacro file that is being loaded. This could be a field in the 3D panel options to add a namespace to a loaded URDF model. This way, we could load the exact same model and only change the namespace in the settings panel instead of the URDF file. Also, in this context, it would be nice to duplicate a URDF in the 3D panel settings so we do not need to import the models one by one.

Please let me know if I should give more specifications or if there are any other questions.

jtbandes, 2022-11-29:
How are namespaces specified in the URDF file? Can you give an example?

Is this a feature supported by RViz?

Dave-van-der-Meer, 2022-11-30:
> Is this a feature supported by RViz?

This feature is supported by RViz2 (and I also think RViz) when listening to the robot_description topic of the robots. When selecting the topic `/leo01/robot_description`, it will automatically take the namespace for the links since it is published on the topic as `leo01/base_link`. In case the topic is not using the namespace, but the links do, you can specify it with the field `TF prefix` in the field below the topic name. Then, the topic might be `/robot_description`, and the links will be `leo01/base_link`.

> How are namespaces specified in the URDF file? Can you give an example?

In the working example, the namespace is given in the URDF file which I converted from the urdf.xacro file. It is hard coded as follows:

```xml
<link name="leo01/base_link">
    <visual>
    <origin xyz="0.0 0.0 0.0" rpy="0 0 0"/>
      <geometry>
        <mesh filename="/home/dave/ros2_ws/leorover_description/models/model_extensions/long_camera_mast.dae"/>
      </geometry>
    </visual>
    <inertial>
      <origin xyz="0.0 0.0 0.0"/>
      <mass value="0.25"/>
      <inertia
        ixx="0.002956" ixy="-0.000001489324" ixz="-0.000008103407"
        iyy="0.02924"  iyz="0.00007112"
        izz="0.02832"/>
      <geometry>
        <cylinder length="0.2" radius="0.025"/>
      </geometry>
    </inertial>
  </link>

  <joint name="chassis_joint" type="fixed">
    <parent link="leo01/base_link"/>
    <child link="leo01/chassis"/>
    <origin xyz="0.09 0.0 -0.001" rpy="0 0 0"/>
  </joint>
```

And in the urdf.xacro file, it is specified as follows:

```xml

<xacro:macro name="leo_mast" 
             params="default_mikrotik_antenna:=true
                     d455_camera:=true
                     link_prefix:=''
                     joint_prefix:=''">

<link name="${link_prefix}mast_link">
    <visual>
    <origin xyz="0.0 0.0 0.0" rpy="0 0 0"/>
      <geometry>
        <mesh filename="package://leorover_description/models/model_extensions/long_camera_mast.dae"/>
      </geometry>
    </visual>
    <inertial>
      <origin xyz="0.0 0.0 0.0"/>
      <mass value="0.25"/>
      <inertia
        ixx="0.002956" ixy="-0.000001489324" ixz="-0.000008103407"
        iyy="0.02924"  iyz="0.00007112"
        izz="0.02832"/>
      <geometry>
        <cylinder length="0.2" radius="0.025"/>
      </geometry>
    </inertial>
  </link>

  <joint name="mast_joint" type="fixed">
    <parent link="${link_prefix}base_link"/>
    <child link="${link_prefix}mast_link"/>
    <origin xyz="0.09 0.0 -0.001" rpy="0 0 0"/>
  </joint>
```

So, in the end, the `link_prefix:=''` is defined inside the launch file and given as a parameter. This then starts the state_publisher to add the link_prefix specified. In our case, it is leo01.

In the 3D panel, I could not find the topic robot_description, so I loaded it from a file in Foxglove. When I load the urdf_xacro file, it will actually display the robot model, but without namespace and not attached to the robot's TF. So there will be the entire TF tree from the model and also the TF tree from the state_publisher with namespace. So a feature to specify the namespace in the robot model file that could add it for the links would be really nice.