foxglove/studio #4917: Extension panels repeatedly seeing didSeek=true when connected to a data source that does not support seeking (ws-protocol)

jhurliman, 2022-12-01:
1. Connect to `foxglove_bridge` and subscribe to at least two topics, one lightweight and one heavyweight (such as an image stream).
2. Observe 3D panel and Image panel repeatedly flashing/strobing

This is due to many frames coming in with `didSeek=true`. Any time a message is received on the WebSocket with a message timestamp less than the previous message timestamp, this happens:

```typescript
// If time goes backwards, increment lastSeekTime and discard unemitted messages from before
// the discontinuity. This prevents us from queueing an unbounded number of messages when
// servers loop over the same recorded data multiple times. However, for now the queue can
// still grow unboundedly in a live system if the listener is not processing messages (such
// as when the app is hidden/backgrounded).
if (this._currentTime && isLessThan(receiveTime, this._currentTime)) {
  ++this._lastSeekTime;
  this._parsedMessages = [];
}
```

defunctzombie, 2022-12-01:
John: "I can tell you, I didn't seek. That's why it's marked as a bug"

jhurliman, 2022-12-01:
Here's a visual demonstration:

https://user-images.githubusercontent.com/195374/205184152-bc0673b5-393d-4e65-b188-83867e74c389.mp4

jtbandes, 2022-12-02:
Either duplicate of, or just blocked by, https://github.com/foxglove/studio/issues/3194

PR #5057 (2023-01-03): Foxglove websocket: Use system time if server does not publish time messages
**User-Facing Changes**
- Foxglove websocket: Use system time if server does not publish time messages

**Description**
*For some reason #5012 got automatically closed and I can't reopen it. This PR replaces it*

This PR is based on #5013 and only to be merged after 1-2 releases after the release that contains #5013. It is the second part of the migration strategy proposed in https://github.com/foxglove/studio/pull/4984#issuecomment-1360366496

From #5013:

> Since https://github.com/foxglove/ws-protocol/pull/299, the [... cut]
commit f386c692 (2023-01-03): Foxglove websocket: Use system time if server does not publish time messages (#5057)

**User-Facing Changes**
- Foxglove websocket: Use system time if server does not publish time
messages

**Description**
*For some reason #5012 got automatically closed and I can't reopen it.
This PR replaces it*

This PR is based on #5013 and only to be merged after 1-2 releases after
the release that contains #5013. It is the second part of the migration
strategy proposed in
https://github.com/foxglove/studio/pull/4984#issuecomment-1360366496

From #5013:

> Since https://github.com/foxglove/ws-protocol/pull [... cut]
closed 2023-01-03 by jhurliman