foxglove/studio #4921: Studio gets slow after an hour of monitoring + recording?

mhl787156, 2022-12-02:
**Description**
Hi,

Foxglove has been really useful for me so far! Especially the magical update which allowed me to see custom msgs over a standard ROS2 connection. 

When I work, I tend to just leave Foxglove on in the background and after about an hour the interface tends to get a bit laggy and slow. I mostly use the default plots so I guess its just collected a large amount of data. 

[image]

I would quite like to keep the data around as its useful for debugging, so I was wondering if you had any suggestions of long running monitoring? Right now I just press F5 every hour! 

I think it would be cool if Foxglove has a recording utility in the interface so that the interface only shows the last n seconds/minutes with the rest being placed into an mcap or something, while allowing scrubbing back and forth at any time. 

- Version: 1.31.0
- OS: Ubuntu 20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS2 native
- ROS distro (if applicable): Foxy

**Steps To Reproduce**

Leave foxy recording a number of 30+hz streams for an hour.

**Expected Behavior**
The UI for the plots is still responsive to scrubbing and navigating around

**Actual Behavior**
Gets a little bit laggy

jtbandes, 2022-12-02:
Have you tried configuring the X Range setting for the Plot panels so that they don't accumulate data indefinitely?

[image]

mhl787156, 2022-12-13:
Hi @jtbandes apologies, yes I tried the range setting! I usually set it to 300 seconds as I've noticed things start lagging if its set to 600 seconds. 

Main request I suppose is to be able to somehow pan to historical data and still see something outside of the collected range. I would also request a pause button as sometimes the data dissapears as I'm analysing/checking it! 

Context is I'm collecting/want to view/ want to check data for 10 minute runs of an experiment. I do collect bags, but its a faff to convert them if I only want to check everything is okay. So I have a live connection of Foxglove, but ofc I only see the last 5 minutes before it slowly dissapears. So having an in-between solution would be really cool.

jtbandes, 2022-12-16:
Yep, we have some prior discussion on this topic: https://github.com/orgs/foxglove/discussions/220

closed 2022-12-16 by jtbandes