foxglove/studio #4935: docker build fails on yarn install

therishidesai, 2022-12-06:
**Description**
When running `docker build` I get the following error:

```
Step 5/12 : RUN yarn install --immutable
 ---> Running in ba9774b51095
Syntax Error: Unexpected identifier
(function (exports, require, module, __filename, __dirname) { version https://git-lfs.github.com/spec/v1
                                                                      ^^^^^

SyntaxError: Unexpected identifier
    at new Script (node:vm:100:7)
    at NativeCompileCache._moduleCompile (/usr/local/lib/node_modules/corepack/dist/vcc.js:251:18)
    at Module._compile (/usr/local/lib/node_modules/corepack/dist/vcc.js:195:36)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1209:10)
    at Module.load (node:internal/modules/cjs/loader:1033:32)
    at Function.Module._load (node:internal/modules/cjs/loader:868:12)
    at Module.require (node:internal/modules/cjs/loader:1057:19)
    at require (/usr/local/lib/node_modules/corepack/dist/vcc.js:170:20)
    at UH (/root/.cache/node/corepack/yarn/3.1.0/yarn.js:382:4931)
    at mu (/root/.cache/node/corepack/yarn/3.1.0/yarn.js:382:5785)
The command '/bin/sh -c yarn install --immutable' returned a non-zero code: 1
```

- Version: latest master
- OS:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
- ROS distro (if applicable):

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

therishidesai, 2022-12-06:
This may be a related issue: [https://github.com/yarnpkg/yarn/issues/1337](https://github.com/yarnpkg/yarn/issues/1337)

jtbandes, 2022-12-06:
You'll need to install Git LFS on your system and run `git lfs pull`. Since the Dockerfile copies files into the container from your local filesystem, the files have to be fetched before `docker build`.

therishidesai, 2022-12-06:
That was it

closed 2022-12-06 by jtbandes