foxglove/studio #4937: Remote rosbag in url gives "Unknown data source: ros1-remote-bagfile"

RileyEv, 2022-12-06:
**Description**
Since v1.34.0 remote rosbags do not open giving message "Unknown data source: ros1-remote-bagfile"

- Version: 1.34.0
- OS: docker - ghcr.io/foxglove/studio:1.34
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ros1-remote-bagfile

**Steps To Reproduce**
Loading bag with url, for example using the demo dataset as a rosbag https://studio.foxglove.dev/?ds=ros1-remote-bagfile&ds.url=https%3A%2F%2Fassets.foxglove.dev%2Fdemo.bag&layoutId=07c2457d-9a4d-48b1-888e-e2e21177737f

**Expected Behavior**
As like in v1.33.0 the bag should load without issue

**Actual Behavior**
In v1.34.0 the bag does not load and gives this message 
[image]

jtbandes, 2022-12-06:
Thanks for the report. It seems like the url param changed, so it needs to be `ds=remote-file` now. We'll look at fixing so the old urls continue to work.

RileyEv, 2022-12-06:
Ahh, it might be worth updating the docs to reflect and maybe a note in the release notes.

defunctzombie, 2022-12-07:
Regressed in: https://github.com/foxglove/studio/pull/4928

commit 4714d359 (2022-12-07): Add `legacyIds?` to IDataSourceFactory (#4939)

**Description**
When we combined the Ros1Remote and McapRemote factories into a single
factory we lost their respective factory `id` fields. These fields are
used in deep-link urls to specify which factory handles the rest of the
query args. This change broke existing URLs using these ids.

This fixes the broken behavior by introducing a new optional `legacyIds`
field to IDataSourceFactory and updating the selectSource(id) function
in PlayerManager to lookup into these ids when selecting a source. Users
are able to use the old ids and the new id  [... cut]
closed 2022-12-07 by defunctzombie