foxglove/studio #4945: Error with yarn install

mandicLuka, 2022-12-08:
**Description**
I am trying to run foxglove studio from source, but I got this error while installation:

```
#12 3.145 Syntax Error: Unexpected identifier                                                                                                                                                
#12 3.145 (function (exports, require, module, __filename, __dirname) { version https://git-lfs.github.com/spec/v1                                                                           
#12 3.145                                                                       ^^^^^                                                                                                        
#12 3.145 
#12 3.145 SyntaxError: Unexpected identifier
#12 3.145     at new Script (node:vm:100:7)
#12 3.145     at NativeCompileCache._moduleCompile (/usr/local/lib/node_modules/corepack/dist/vcc.js:251:18)
#12 3.145     at Module._compile (/usr/local/lib/node_modules/corepack/dist/vcc.js:195:36)
#12 3.145     at Object.Module._extensions..js (node:internal/modules/cjs/loader:1209:10)
#12 3.145     at Module.load (node:internal/modules/cjs/loader:1033:32)
#12 3.145     at Function.Module._load (node:internal/modules/cjs/loader:868:12)
#12 3.145     at Module.require (node:internal/modules/cjs/loader:1057:19)
#12 3.145     at require (/usr/local/lib/node_modules/corepack/dist/vcc.js:170:20)
#12 3.145     at UH (/root/.cache/node/corepack/yarn/3.1.0/yarn.js:382:4931)
#12 3.145     at mu (/root/.cache/node/corepack/yarn/3.1.0/yarn.js:382:5785)
```

I even tried using Docker, same thing.

- Version: 1.34.1, 1.33.0
- OS: WSL2
- node version: v19.2.0
- git-lfs version: 3.3.0
- yarn version: 1.22.19

**Steps To Reproduce**
Clone repo
Run corepack enable && yarn install

jtbandes, 2022-12-08:
Sounds like you need to run `git lfs pull`.

defunctzombie, 2022-12-13:
Closing this since a possible solution has been provided. @mandicLuka If you run the recommended command but continue to experience issues please re-open and provide more information on how to reproduce.

closed 2022-12-13 by defunctzombie