foxglove/studio #4959: Cannot publish /clicked_point msg with ROS 1 native connection

qdhuxp, 2022-12-11:
**Description**
When connect foxglove to ROS1(noetic) with **http://xxx:11311**,  foxglove cannot publish **/clicked_point** msg to ROS Master. But if connect foxglove to ROS1(noetic) with rosbridge, it works.  In Ros master machine, you can use "**rostopic echo /clicked_point**" to monitor the received messages.

- Version: latest
- OS: Ubuntu/windows
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): **ROS 1 native**
- ROS distro (if applicable): **noetic**

**Steps To Reproduce**

1. Connect foxglove to ROS1(noetic) native with http://xxx:11311;
2. In Ros Master machine, use "rostopic echo /clicked_point" to monitor the received messages;
3. In 3D panel, use publish function to publish /clicked_point msg to ROS Master;

**Expected Behavior**
Ros master echo command should print /clicked_point messages.

**Actual Behavior**
Ros master echo command cannot print anything.

foxymiles, 2022-12-12:
This sounds like it might be a networking issue. Can you please double check that the hostname you entered for `ROS_HOSTNAME` in this connection dialog is reachable from your ros nodes?

[image]

qdhuxp, 2022-12-19:
Thanks @foxymiles , but I am sure I use the correct ROS_HOSTNAME, otherwise, foxglove will not receive msgs from Ros master at all, right? BTW, I use ROS_IP and ROS_MASTER_URI in .bashrc to make foxglove connectable to Ros master. 
[image]

foxymiles, 2022-12-22:
Thanks for the extra information @qdhuxp . Can you please try publishing from a teleop panel and see if that works in your setup?

jtbandes, 2023-01-06:
Closing out this ticket for now since we haven't received any further information.

closed 2023-01-06 by jtbandes