foxglove/studio #4967: Unable to load local extension when org extension installed/cached

jtbandes, 2022-12-12:
hello again! has anyone experienced caching old code after publishing an extension, or know how I can load new local updates to an extension? Here is what I tried:
1. publish an extension to org
2. make new (local) changes to extension
3. attempt to load new local code by running `yarn local-install` + refresh Studio. (old code (version published to org) is still running when I open the [local] version)
4. try bumping `package.json` version, deleting everything in `.foxglove-studio/extensions/` , log out, quit and reopen Foxglove Studio app, and retry step 3 but changes are still not loaded
before this, yarn local-install + refresh studio had worked to reload changes, so not sure what happened, or if I’m missing a step. thank you in advance!

[Slack Message](https://foxglove.slack.com/archives/C028UEY858S/p1670873520807189?thread_ts=1670873520.807189&cid=C028UEY858S)

defunctzombie, 2022-12-13:
FG-934

defunctzombie, 2022-12-13:
FG-934

PR #5072 (2023-01-04): Fix extension id collisions across multiple loaders
**User-Facing Changes**
Fixed a bug that prevented the use of org-wide extensions and locally-installed extensions with the same ID.

**Description**
We were calling `getExtensions()` and then calling `loadExtension(id)` on a different loader.
This bug seems to have been present since the introduction of multiple loaders (https://github.com/foxglove/studio/pull/3836).

[View diff ignoring whitespace](https://github.com/foxglove/studio/pull/5072/files?diff=split&w=1)

Fixes #4967 
Fixes FG-934
commit 4e707a8a (2023-01-04): Fix extension id collisions across multiple loaders (#5072)

**User-Facing Changes**
Fixed a bug that prevented the use of org-wide extensions and
locally-installed extensions with the same ID.

**Description**
We were calling `getExtensions()` and then calling `loadExtension(id)`
on a different loader.
This bug seems to have been present since the introduction of multiple
loaders (https://github.com/foxglove/studio/pull/3836).

Fixes #4967 
Fixes FG-934
closed 2023-01-04 by jtbandes