foxglove/studio #4985: Opening a layout with certain panels immediately marks the layout as "unsaved"

defunctzombie, 2022-12-15:
I've made several layouts in studio with a variety of panels. When I switch to some of these layouts they immediately get marked as unsaved even tho I've not made any changes.

After tracing down some differences in the configs before/after here are some items which are changed:

```
map.center = undefined
map.zoomLevel = undefined
RawMessages.expansion = undefined
3D.followTf = undefined
StateTransition.isSynced = true
Text.fontSize = 0.7
Plot.title = 'Plot'
```

The ideal behavior is that opening a layout does not immediately mark it as dirty.

foxymiles, 2022-12-15:
I'm not seeing this at least with simple 1-panel layouts. Any more detailed steps to repro or a specific layout that demonstrates the issue?

https://user-images.githubusercontent.com/93935560/207973493-5d3c7dfd-4c89-43e7-bc1d-f077b029eae7.mov

defunctzombie, 2022-12-16:
For the map panel - can you load some data and then maybe try opening the layout with and without data after having had data? Thinking about what might cause _center_ to be set vs not set.

Have you tried the 3d panel? State transition panel?

foxymiles, 2022-12-21:
The sample layout is sort of a special case because of the way we handle it. I'll try to repro with a custom layout.

jtbandes, 2023-01-04:
We can update the sample layout so it doesn't trigger this problem and close out this ticket.

PR #5106 (2023-01-06): sample layout: add explicit followTf
**User-Facing Changes**
Not worth calling out in release notes

**Description**
Fixed an issue where the sample layout would appear modified immediately after it was created. The 3D panel would trigger a save with `followTf: undefined`, which would result in the same layout after JSON.stringify, but is considered unequal by this equality check:

https://github.com/foxglove/studio/blob/ecf5c4b2a20937f7ba115acc444878e7bb69f848/packages/studio-base/src/providers/CurrentLayoutProvider/index.tsx#L173-L177

Fixes #4985 
Fixes FG-996
commit e52414e5 (2023-01-06): sample layout: add explicit followTf (#5106)

**User-Facing Changes**
Not worth calling out in release notes

**Description**
Fixed an issue where the sample layout would appear modified immediately
after it was created. The 3D panel would trigger a save with `followTf:
undefined`, which would result in the same layout after JSON.stringify,
but is considered unequal by this equality check:

https://github.com/foxglove/studio/blob/ecf5c4b2a20937f7ba115acc444878e7bb69f848/packages/studio-base/src/providers/CurrentLayoutProvider/index.tsx#L173-L177

Fixes #4985 
Fixes FG-996
closed 2023-01-06 by jtbandes