foxglove/studio #4989: User script changes do not update until another message arrives

defunctzombie, 2022-12-16:
**Description**
I've created a layout with two raw message panels and a user script panel. I've written a basic user script using the default "new script" template which outputs `hello: world`. My script has one input topic and one output topic. One raw message panel is subscribed to my input topic and one to my output topic.

I now load a data source that contains messages on my input topic. I play through the data some and pause. The raw message panel shows me the expected output `hello: world`. I update the user script to change the string `world` to something else and click `save`. My output raw message panel does not update with the new message. If I start to play the output raw message panel finally updates.

**Expected Behavior**
The output raw message panel updates with the new message. I believe this behavior use to work (it does seem to work with webviz).

Maybe we've changed something in our user script panel or in our message pipeline that does not allow us to re-transmit the same "frame" again but it is worth investigating why we regressed.

jtbandes, 2022-12-16:
It appears that webviz was triggering a backfill when variables changed in the OrderedStampPlayer:

https://github.com/cruise-automation/webviz/blob/4ed917d47e57d4286d7aa4b5c783e3f26420446e/packages/webviz-core/src/players/OrderedStampPlayer.js#L183-L188

jtbandes, 2022-12-16:
Related: https://github.com/foxglove/studio/issues/1910

PR #5071 (2023-01-06): Saving User scripts updates subscribed panels
**User-Facing Changes**

- Saving a user script will update subscribed panels with new output messages

**Description**

 Added state to remember last of each topic's message in the userNodePlayer and every time `setUserNodes` is called as an update to an existing node, that node will be added to a set of nodes that need to have their outputs refreshed. In the next `_onPlayerState` call this will read in the previous messages for each input topic and add them to the messages to be processed in `_getMessages` and then remove them from the list of no [... cut]
commit 01667857 (2023-01-06): Saving User scripts updates subscribed panels (#5071)

**User-Facing Changes**

- Saving a user script will update subscribed panels with new output
messages

**Description**

Added state to remember last of each topic's message in the
userNodePlayer and every time `setUserNodes` is called as an update to
an existing node, that node will be added to a set of nodes that need to
have their outputs refreshed. In the next `_onPlayerState` call this
will read in the previous messages for each input topic and add them to
the messages to be processed in `_getMessages` and then remove them from
the li [... cut]
closed 2023-01-06 by snosenzo