foxglove/studio #4990: React conflict after create a customer panel

ShuoShenDe, 2022-12-19:
**Description**
I want to create custom panel, so I follow the [[guidance](https://foxglove.dev/docs/studio/extensions/guide-create-custom-panel)]

After I run the command under desktop folder:
`npm init foxglove-extension@latest myExtensionName`

I will get the error when I restart the desktop app.

- Version:
node -v: 16.15.1
npm -v: 8.11.0

- Platform:
Ubuntu 18
**Steps To Reproduce**

[image]

**Expected Behavior**

ShuoShenDe, 2022-12-19:
When I run  `npm ls react`, I can get the info as follows:

 ```
(base) ubuntu@i-0bb5835e3202d67a4:~/Documents/NVME/Web-PC-Stream/desktop$ npm ls react
foxglove-studio@1.35.0-dev /home/ubuntu/Documents/NVME/Web-PC-Stream
└─┬ desktop@ -> ./desktop
  ├─┬ @foxglove/studio-base@1.35.0-dev invalid: "workspace:*" from desktop -> ./packages/studio-base
  │ ├─┬ @emotion/react@11.7.1
  │ │ └── react@17.0.2 deduped
  │ ├─┬ @emotion/styled@11.6.0
  │ │ └── react@17.0.2 deduped
  │ ├─┬ @fluentui/react-icons-mdl2@1.2.14
  │ │ ├─┬ @fluentui/react-icon-provider@1.2.13
  │ │ │ ├─┬ @fluentui/style-utilities@8.5.4
  │ │ │ │ └─┬ @fluentui/theme@2.4.8
  │ │ │ │   └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized, "^16.13" from node_modules/react-dnd-multi-backend, "^16.13.1" from node_modules/react-dnd-preview
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized
  │ │ ├─┬ @fluentui/utilities@8.4.0
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized
  │ │ └── react@17.0.2 deduped
  │ ├─┬ @foxglove/hooks@ invalid: "workspace:*" from packages/studio-base -> ./packages/hooks
  │ │ └── react@17.0.2 deduped
  │ ├─┬ @mui/icons-material@5.4.1
  │ │ └── react@17.0.2 deduped
  │ ├─┬ @mui/material@5.4.1
  │ │ ├─┬ @mui/base@5.0.0-alpha.68
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized
  │ │ ├─┬ @mui/system@5.4.1
  │ │ │ ├─┬ @mui/private-theming@5.4.1
  │ │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized, "^16.13" from node_modules/react-dnd-multi-backend
  │ │ │ ├─┬ @mui/styled-engine@5.4.1
  │ │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized, "^16.13" from node_modules/react-dnd-multi-backend
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized
  │ │ ├─┬ @mui/utils@5.4.1
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized
  │ │ └── react@17.0.2 deduped
  │ ├─┬ @storybook/addon-actions@6.5.10
  │ │ ├─┬ @storybook/addons@6.5.10
  │ │ │ ├─┬ @storybook/router@6.5.10
  │ │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized, "^16.13" from node_modules/react-dnd-multi-backend
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized
  │ │ ├─┬ @storybook/api@6.5.10
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized
  │ │ ├─┬ @storybook/components@6.5.10
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized
  │ │ ├─┬ @storybook/theming@6.5.10
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized
  │ │ ├─┬ react-inspector@5.1.1
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtualized
  │ │ └── react@17.0.2 deduped
  │ ├─┬ @storybook/react@6.5.10
  │ │ ├─┬ @storybook/core-common@6.5.10
  │ │ │ └── react@17.0.2 deduped invalid: "^0.14.7 || ^15.0.0-0 || ^16.0.0-0" from node_modules/react-autocomplete, "~0.14.9 || ^15.0.0 || ^16.0.0" from node_modules/react-hover-observer, "^15.3.0 || ^16.0.0-alpha" from node_modules/react-virtu [... cut ...]
  └─┬ tss-react@3.7.1
    └── react@18.2.0 deduped invalid: "17.0.2" from desktop/node_modules/react-dom, "^16.8.0  || ^17.0.0" from desktop/node_modules/react-use

npm ERR! code ELSPROBLEMS
npm ERR! invalid: react@18.2.0 /home/ubuntu/Documents/NVME/Web-PC-Stream/desktop/node_modules/react
npm ERR! invalid: react@17.0.2 /home/ubuntu/Documents/NVME/Web-PC-Stream/node_modules/react

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ubuntu/.npm/_logs/2022-12-19T12_50_48_731Z-debug-0.log
```

snosenzo, 2022-12-19:
I'm not sure where your extension is located but it looks like the submodule in `Web-PC-Stream/desktop` uses `react@18.2.0` which is causing npm to reconcile at the `Web-PC-Stream` level to conflict with that version. 

Did you `cd /home/ubuntu/Documents/NVME/Web-PC-Stream/desktop/myExtensionName` and call `yarn local-install`? as seen here? https://foxglove.dev/docs/studio/extensions/local-development

ShuoShenDe, 2022-12-20:
> I wasn't able to reproduce this locally from the guidance steps -- they all seem to be working for me at least on mac. So I'll need a bit more information. I'm not sure where your extension is located but it looks like the submodule in `Web-PC-Stream/desktop` uses `react@18.2.0` which is conflicting with `react@17.0.2` at the `Web-PC-Stream` level. npm dependencies must be consistent across a directory's submodules if you're running at the `Web-PC-Stream` directory.
> 
> Did you `cd /home/ubuntu/Documents/NVME/Web-PC-Stream/desktop/myExtensionName` (if you installed it there) and call `yarn local-install` as seen here? https://foxglove.dev/docs/studio/extensions/local-development

Hi~Dear snosenzo,

I run the `yarn local-install` under my extension folder (show-image), and I got the error as follows:

[image]

ShuoShenDe, 2022-12-20:
My structure of my project is:
`
Web-PC-Stream
├── benchmark
├── ci
├── packages
├── patches
├── resources
├── web
├── desktop
│   ├── show-images(My Example Panel)

`

ShuoShenDe, 2022-12-20:
I reproduce the error on Linux and Windows
1. just clone the studio project
2. Run `corepack enable` and `yarn install` ( no error )
3. Now I can build the desktop and web app correctly, there are some ERROR CONSOLE, but the studio can running correctly
[image]

4. I create the custom panel project `show-image` by running `npm init foxglove-extension@latest show-image` under the `desktop` folder
[image]

5. Then I will get the react conflict error when I run `yarn desktop:start`.
[image]

jtbandes, 2022-12-20:
> My structure of my project is:
> ```
> Web-PC-Stream
> ├── benchmark
> ├── ci
> ├── packages
> ├── patches
> ├── resources
> ├── web
> ├── desktop
> │   ├── show-images(My Example Panel)
> 
> ```

You should execute `npm init foxglove-extension@latest show-image` in a **different** folder outside of `studio`. Extensions are meant to live outside the studio repository, because they can be loaded without modifying the app's code.

snosenzo, 2022-12-20:
Also a couple other things to note:

> `yarn desktop:start`

You need to run `yarn desktop:serve` before `yarn desktop:start` as seen in the [Studio README.md](https://github.com/foxglove/studio)
```
# To launch the desktop app (run both scripts concurrently):
$ yarn desktop:serve        # start webpack
$ yarn desktop:start        # launch electron
```

Also in order to see your local extension in studio be sure to run `yarn local-install` in your extension directory (`show-image`). It doesn't look like you ran it between steps 4 and 5.

ShuoShenDe, 2022-12-21:
> yarn local-install

I run them sequentially.
And if I run `yarn local-install` in my extension directory(show-image), I will got the error:
[image]

jtbandes, 2022-12-21:
Putting the extension repo inside the studio repo is not a supported workflow. Try moving the extension outside of the studio repo.

ShuoShenDe, 2022-12-22:
> Putting the extension repo inside the studio repo is not a supported workflow. Try moving the extension outside of the studio repo.

Thanks for your reply, Now I create the extension out of the studio repo. But how can I debug it? And how my studio find that extension? I'm not sure any tutorial for it?

jtbandes, 2022-12-22:
You need to run `yarn local-install` to install your extension. Check out the readme here: https://github.com/foxglove/create-foxglove-extension#creating-your-first-extension

closed 2022-12-22 by jtbandes