foxglove/studio #4997: Foxglove Studio installer only works when run as Administrator

Aman-Anas, 2022-12-20:
**Description**
Downloaded the x64 windows installer from the releases page (foxglove-studio-1.35.0-win-x64.exe), and tried running it from the downloads folder. A window briefly appeared then disappeared too fast for me to see anything. Tried it several times, no luck. Then I tried running it as administrator, and it worked perfectly fine (it installed and I could launch it). Seems like the installer doesn't run as admin by default.

- Version: 1.35.0
- Platform: Windows 10, 64-bit

**Steps To Reproduce**
1) Download 1.35.0 x64 installer for windows
2) Double-click the installer .exe file (without running as admin)
3) Installation does not occur

**Expected Behavior**
It should automatically run as administrator and request admin rights.

jtbandes, 2022-12-21:
I just tried on Windows 11 with a new non-admin user. I was able to download the installer and run it, and it placed the program into AppData\Local\Programs without requiring admin privileges.

[image]

jtbandes, 2022-12-21:
Got another colleague to test on Windows 10 Pro with a non-admin account — also was able to install and run successfully without admin privileges.

I'm going to close this issue since we are not able to reproduce, but please follow up if you find any reproducible steps to trigger the problem.

closed 2022-12-21 by jtbandes