foxglove/studio #5041: useConfirm doesn't work in dev mode

jtbandes, 2022-12-22:
Double-invoke of effects in Strict Mode causes the prompt to dismiss prematurely due to this code:

https://github.com/foxglove/studio/blob/341f69488fd6fb5e4c2af969966a69fbcafdfd26/packages/studio-base/src/hooks/useConfirm.tsx#L57-L61

The same issue probably affects usePrompt.

It would be better to get rid of ModalHost and set both of these up to return an element that the caller has to render in the react tree themselves.

foxymiles, 2023-01-02:
Do we even need that `useEffect`? I think the MUI dialog will still call `onClose` if the user closes via ESC key.

jtbandes, 2023-01-03:
I think the original reason for adding it was to handle if the parent component unmounts without the user pressing esc.

foxymiles, 2023-01-03:
In that case is it still important to invoke the callback?

closed 2023-01-04 by jtbandes