foxglove/studio #5049: Valid mcap file with nested arrays does not load

defunctzombie, 2022-12-22:
**Description**

I have an mcap file with a valid json schema and message which fails to open in studio but does work with the mcap CLI tool.

[image]

[simple_file.mcap.zip](https://github.com/foxglove/studio/files/10287893/simple_file.mcap.zip)

```
library: python mcap 0.0.16
profile:
messages: 1
duration: 0s
start: 1.000
end: 1.000
compression:
	zstd: [1/1 chunks] [00 MB/00 MB (38.13%)]
channels:
  	(1) matrix_data  1 msgs (+Inf Hz)   : matrix_schema [jsonschema]
attachments: 0
metadata: 0
```

```
(base) ➜  Downloads mcap cat simple_file.mcap --json
{"topic":"matrix_data","sequence":0,"log_time":1.000000000,"publish_time":1.000000000,"data":{"matrix":[[1,0,0]]}}
```

```
  1   matrix_schema  jsonschema  {"$schema": "http://json-schema.org/draft-04/schema#", "type": "object", "properties": {"matrix": {"type": "array", "minItems": 1, "maxItems": 1, "items": {"type": "array", "items": {"type": "integer"}, "minItems": 3, "maxItems": 3}}}}
```

**Expected Behavior**
The mcap file loads without errors and I can see the matrix_data in a raw message panel.

jtbandes, 2022-12-22:
Studio does not and has never supported nested arrays. I think this is a feature request.

defunctzombie, 2022-12-22:
@jtbandes wow yea you are right - I had in my mind that things like the camera info matrix were nested arrays but I see those are single arrays and code is meant to "interpret" them a certain way.

Removing the bug label. Indeed this is a feature request.