foxglove/studio #5051: NetworkError

ShuoShenDe, 2022-12-22:
**Description**
 [image]
My network is good for internet, but the studio gives me an error of NetworkError. The web server has not error logs.
[image]

- Version:
node -v
v16.18.1
npm -v
8.19.2

- Platform:
VSCode
Ubuntu 18

**Steps To Reproduce**

Not sure

**Expected Behavior**
How to find out the problem?

jtbandes, 2022-12-22:
What is the network request that failed?

jtbandes, 2022-12-23:
Moving to a discussion since this seems to be an issue with the development of studio rather than with the production app.