foxglove/studio #5061: Unable to visualize `sensor_msgs/msg/Image` stored in MCAP rosbag

aditya2592, 2022-12-29:
**Description**

I have an MCAP file where an RGB image of type `sensor_msgs/msg/Image` is stored in a topic. When I try to visualize this using the `Image` layout after importing the MCAP file, I get the following error:
```
RangeError: Invalid typed array length: 2682033074
    at new Uint8Array (<anonymous>)
    at CdrReader.uint8Array (file:///snap/foxglove-studio/27/resources/app.asar/renderer/2098.js:1:2766)
    at file:///snap/foxglove-studio/27/resources/app.asar/renderer/2098.js:1:17718
    at MessageReader.readComplexType (file:///snap/foxglove-studio/27/resources/app.asar/renderer/2098.js:1:16780)
    at MessageReader.readMessage (file:///snap/foxglove-studio/27/resources/app.asar/renderer/2098.js:1:15956)
    at Object.deserializer (file:///snap/foxglove-studio/27/resources/app.asar/renderer/1023.js:1:7532)
    at processRecord (file:///snap/foxglove-studio/27/resources/app.asar/renderer/4766.js:1:6970)
    at McapStreamingIterableSource.initialize (file:///snap/foxglove-studio/27/resources/app.asar/renderer/4766.js:1:7255)
    at async McapIterableSource.initialize (file:///snap/foxglove-studio/27/resources/app.asar/renderer/4766.js:1:11355)
    at async WorkerIterableSourceWorker.initialize (file:///snap/foxglove-studio/27/resources/app.asar/renderer/4126.js:1:19821)
```
The same topic when being visualized live through `websocket` works fine. I am able to see the image in Foxglove Studio.

- Version: 1.35.2
- Platform: Ubuntu 22.04

**Steps To Reproduce**

MCAP file was stored with below writer config:

```
noChunkCRC: false
noChunking: false
noMessageIndex: false
noSummary: false
chunkSize: 786432
compression: "Zstd"
compressionLevel: "Fast"
forceCompression: false
```

**Expected Behavior**
Visualize topic of type `sensor_msgs/msg/Image` both live and through the MCAP file

snosenzo, 2023-01-03:
Thanks for filing this -- would you be able to share an example MCAP file that we can test against?

aditya2592, 2023-01-05:
[Here](https://drive.google.com/file/d/195A-jeaFjjP3igY6GjOIziQW7U-HxTz_/view?usp=sharing) is a link to a test bag containing only 1 topic with above error

jtbandes, 2023-01-05:
How did you record this bag? Something has gone wrong in the recording process.

There seems to be a mismatch between the schema name, schema data, and message data. The schema data embedded in the file is the schema for `sensor_msgs/CompressedImage`:

[image]

However, the schema name and the actual message data appear to be a `sensor_msgs/Image` — note the presence of the `width` and `height` (640x480) and the `encoding` (`8UC3`) in the message data:
[image]

[image]

jtbandes, 2023-01-05:
Converting to a discussion because this is not a bug with studio.