foxglove/studio #5069: Can not connect to the target error

ShuoShenDe, 2023-01-04:
**Description**
Hi~Dear foxglove team,
Happy new year. Here I try to debug the studio project. After I clone from GitHub and Run `corepack enable` and `yarn install`. Then I run it with vscode.
[image]
However, it gives me the error as follows:
[image]

What should I do to start debug? Thanks

- Version: 
- node -v
v16.18.1
npm -v
8.19.2
- Platform:
Ubuntu 18

**Steps To Reproduce**

Clone the studio from GitHub
Run `corepack enable` and `yarn install`
Then in vscode, run Attach to Chrome 
**Expected Behavior**
Get the error

ShuoShenDe, 2023-01-04:
I might found the reason, this might beacuse it need chrome-debuggi, so run the command in terminial:
` google-chrome --remote-debuggi-port=9222`
Then you will see a new chrome window. This command can run, but it gives many errors, the breakpoints are still not working.
Is there any tutorial for setting development environment?

defunctzombie, 2023-01-04:
Removing the bug label and moving this to a discussion. We use the bug label for issues with Studio features itself rather than your development environment.