foxglove/studio #5073: The app.asar contains 'quicklook' not 'renderer'!

ChinaPuffer, 2023-01-05:
**Description**
When packaging the studio into  a dmg file, the app.asar file contains the 'quicklook' folder. but in webpack.main.config.ts file the rendererentry is "`file://${require('path').join(__dirname, '..', 'renderer', 'index.html')}`".

- Version:lastest
- Platform:mac os

**Steps To Reproduce**

**Expected Behavior**

jtbandes, 2023-01-05:
Please explain the issue in more detail. What problem are you experiencing? Are you having trouble using the .dmg app package downloaded from the foxglove website or GitHub releases?

ChinaPuffer, 2023-01-05:
The .dmg app package downloaded from the GitHub releases runs ok! But i use "yarn package:darwin" to package the studio, the app.asar file does not contain the 'renderer' folder! How can i fix this problem!

ChinaPuffer, 2023-01-05:
> Please explain the issue in more detail. What problem are you experiencing? Are you having trouble using the .dmg app package downloaded from the foxglove website or GitHub releases?

What steps do i need to take before i use the "yarn package:darwin" to package it?

ChinaPuffer, 2023-01-05:
> > Please explain the issue in more detail. What problem are you experiencing? Are you having trouble using the .dmg app package downloaded from the foxglove website or GitHub releases?
> 
> What steps do i need to take before i use the "yarn package:darwin" to package it?

I found the problem: it was caused by some unused declarations!
When some mistakes like "'LayoutBrowser' is declared but its value is never read." appear, the webpack tool will not generate the 'renderer' folder!

closed 2023-01-05 by ChinaPuffer