foxglove/studio #5158: Error during the installation process

iSamBa, 2023-01-17:
**Description**
After enabling `corepack` and during the installation process `yarn install` I get the following error :

```
Syntax Error: Unexpected identifier 'https'
(function (exports, require, module, __filename, __dirname) { version https://git-lfs.github.com/spec/v1
                                                                      ^^^^^
SyntaxError: Unexpected identifier 'https'
    at new Script (node:vm:100:7)
    at NativeCompileCache._moduleCompile (/opt/homebrew/Cellar/corepack/0.15.3/libexec/lib/node_modules/corepack/dist/vcc.js:251:18)
    at Module._compile (/opt/homebrew/Cellar/corepack/0.15.3/libexec/lib/node_modules/corepack/dist/vcc.js:195:36)
    at Module._extensions..js (node:internal/modules/cjs/loader:1293:10)
    at Module.load (node:internal/modules/cjs/loader:1096:32)
    at Module._load (node:internal/modules/cjs/loader:935:12)
    at Module.require (node:internal/modules/cjs/loader:1120:19)
    at require (/opt/homebrew/Cellar/corepack/0.15.3/libexec/lib/node_modules/corepack/dist/vcc.js:170:20)
    at UH (/Users/aissam/.cache/node/corepack/yarn/3.1.0/yarn.js:382:4931)
    at mu (/Users/aissam/.cache/node/corepack/yarn/3.1.0/yarn.js:382:5785)
```
The same error happens while building the docker image using the provided `Dockerfile`

````
=> ERROR [build 5/6] RUN yarn install --immutable                                                                                                                                                                     3.0s
------                                                                                                                                                                                                                      
 > [build 5/6] RUN yarn install --immutable:                                                                                                                                                                                
#12 2.944 Syntax Error: Unexpected identifier                                                                                                                                                                               
#12 2.944 (function (exports, require, module, __filename, __dirname) { version https://git-lfs.github.com/spec/v1
#12 2.944                                                                       ^^^^^
#12 2.944 
#12 2.944 SyntaxError: Unexpected identifier
#12 2.944     at new Script (node:vm:100:7)
#12 2.944     at NativeCompileCache._moduleCompile (/usr/local/lib/node_modules/corepack/dist/vcc.js:251:18)
#12 2.944     at Module._compile (/usr/local/lib/node_modules/corepack/dist/vcc.js:195:36)
#12 2.944     at Object.Module._extensions..js (node:internal/modules/cjs/loader:1219:10)
#12 2.944     at Module.load (node:internal/modules/cjs/loader:1043:32)
#12 2.944     at Function.Module._load (node:internal/modules/cjs/loader:878:12)
#12 2.944     at Module.require (node:internal/modules/cjs/loader:1067:19)
#12 2.944     at require (/usr/local/lib/node_modules/corepack/dist/vcc.js:170:20)
#12 2.944     at UH (/root/.cache/node/corepack/yarn/3.1.0/yarn.js:382:4931)
#12 2.944     at mu (/root/.cache/node/corepack/yarn/3.1.0/yarn.js:382:5785)
------
executor failed running [/bin/sh -c yarn install --immutable]: exit code: 1
````
- Version: `master`
- Platform: `MacOs 13.1`

**Steps To Reproduce**

1. clone the repository

2. run `corepack enable` then `yarn install`

***Alternatively:***
1. clone the repository
2. run `docker build .`

jtbandes, 2023-01-17:
You need to install Git LFS before cloning the repo.

Duplicate of https://github.com/foxglove/studio/issues/4935, https://github.com/foxglove/studio/issues/4945

closed 2023-01-17 by jtbandes

jaguardo, 2023-04-12:
```
git lfs install
git clone git@github.com:foxglove/studio.git
git lfs pull
```