foxglove/studio #5270: Publish pose estimate button in 3D panel does not appear with foxglove-bridge

MarkSisin, 2023-01-31:
To perform re-localization of the vehicle, we want to use the Foxglove 3D panel with foxglove-bridge ROS node to publish the pose estimate from the point cloud. We know that this functionality is implemented in Foxglove Studio, and there are even Publish options in configuration of 3D panel.
We made foxglove-bridge choice in favor of rosbridge for performance reasons. However, it seems that either foxglove-bridge contains limitations in terms of data it can accept from Studio, or it is a bug in the 3D panel component. Unfortunately, we can not send the pose estimate from Foxglove Studio to ROS via foxglove-bridge, because it does not show the button in panel's UI.
 
- Version: 1.39
- Platform: 
  1. Ubuntu 18.04. ROS1 Melodic. (We launch ROS stack in Docker container)
  2. Ubuntu 22.04. Foxglove docker image with tag 1.39
  3. ros-melodic-foxglove-bridge/bionic,now 0.2.2-1bionic.20221213.131445 amd64. ROS1 melodic foxglove bridge node.

**Steps To Reproduce**
As a reproduction scenario, I establish foxglove-bridge WS connection from my ROS application to Web version of Foxglove.
I start publishing sensor_msgs/PointCloud2 information and can see the generated point cloud in 3D panel.
However, I can not see the button that allows me to publish Pose estimate from Foxglove back to ROS.

If I use rosbridge WS connection with my ROS application, then I can see both generated point cloud in 3D panel and the Pose estimate button with _data-testid=publish-button_. Thus, I can not reproduce the problem with rosbridge.

**Expected Behavior**
Pose estimate button is available in 3D panel with foxglove-bridge, similarly to rosbridge.

achim-k, 2023-02-14:
Unfortunately, this doesn't work with the current release of `foxglove-bridge` as https://github.com/foxglove/ros-foxglove-bridge/pull/134 is missing. I will create a new release today, but it might take some time for the new melodic sync to happen. Until then, is running `foxglove-bridge` from source an option for you?

---

Publishing via the 3D panel is only possible if the datasource declares that it is a ros1/ros2 source:

https://github.com/foxglove/studio/blob/23edfbf0188a24daa09bc3838e9d0ae869068fa2/packages/studio-base/src/panels/ThreeDeeRender/ThreeDeeRender.tsx#L1019-L1022

The foxglove websocket datasource does this since #5319 (which relies on https://github.com/foxglove/ros-foxglove-bridge/pull/134).

jtbandes, 2023-02-21:
To clarify, the functionality is already implemented, and should work if you build the foxglove_bridge from source. But yes, you will have to wait if you want to use it from the official release channels.

achim-k, 2023-02-23:
Closing as this will be solved with the next melodic sync when `foxglove_bridge` version `0.4.1` will be made available.

closed 2023-02-23 by achim-k