foxglove/studio #5286: GeoJSON Point not used as 'Follow' Topic on Map

brt-jeffrey-beaupre, 2023-02-02:
**Description**
I have a message converter than generates a single point GeoJson message. The point shows up on the map, but when I use it as the Follow topic, the Map blinks and fails to "follow" the point.

- Version: 1.36.0
- Platform: Ubuntu

**Steps To Reproduce**

Create a message converter to generate a single point geojson message.

**Expected Behavior**
Follow the geojson point just like a LocationFix message.

achim-k, 2023-02-02:
Seems related to #5135 which got fixed in `v1.37.0`. Could you upgrade Studio or use studio.foxglove.dev and confirm that the issue is gone?

brt-jeffrey-beaupre, 2023-02-02:
I can confirm in 1.39.1 that it doesn't flash ... but it still doesnt seem to follow the topic. I can toggle to another topic with LocationFix messages and the view jumps over and follows, but if i switch back to the geojson topic it does not move back.

foxymiles, 2023-02-03:
That's correct the follow feature doesn't currently work on GeoJSON messages:

https://github.com/foxglove/studio/blob/d5cd6c057b4c81109967241ccec1b29e926cf57f/packages/studio-base/src/panels/Map/MapPanel.tsx#L479-L505

I think for this to work we'd have to add some logic to estimate a center from a GeoJSON message since it can span an arbitrary space.

foxymiles, 2023-02-03:
At a minimum we should probably exclude GeoJSON topics from the follow list until we do implement this.

foxymiles, 2023-02-07:
@brt-jeffrey-beaupre For now I've just removed the GeoJSON topics from the follow list since we don't actually currently support following GeoJSON topics. If you'd like to see this added as a new feature please open a discussion here:

https://github.com/orgs/foxglove/discussions/categories/studio

closed 2023-02-07 by foxymiles