foxglove/studio #5306: Failed to initialize camera model from CameraInfo

gilduz, 2023-02-08:
**Description**
I'm unable to transform image markers in the image panel. 
Image camera info have distortion_model": "plumb_bob".

- Version: 1.39.1
- Platform: Windows 10, connected to ROS2 via foxglove_bridge

**Steps To Reproduce**

1. Publish ROS2 images sensor_msgs/msg/Image and respective sinchronized camera info (distortion_model": "plumb_bob")
2. Publish ROS2 image markers visualization_msgs/msg/ImageMarker, synchronized with images timestamp
3. Markers are correctly overlayed to images if "Transform markers" is disabled
4. Markers are not overlayed to images if "Transform markers" is enabled (error message: Failed to initialize camera model from CameraInfo. Details: TypeError: Cannot read properties of undefined (reading '0'))

**Expected Behavior**
Markers are correcly overlayed to images if "Transform markers" is enabled

jtbandes, 2023-02-13:
Can you please share an example mcap file that contains example data? This will make it much easier to reproduce the issue and work on a fix.

jtbandes, 2023-03-03:
Hi, I'm going to close out this issue since we haven't heard back for a while. If you're able to share more information (ideally sample data or scripts to reproduce the issue) don't hesitate to reply and we can reopen the issue.

closed 2023-03-03 by jtbandes